supermerill/SuperSlicer #4141: 2.5.59.8 seems to ignore speed/acceleration limits

ABLomas, 2024-02-19:
### What happened?

My printer is not fastest so i had preset limits (in marlin and in superslicer) as 200mm/s max for speeds.
Latest (2.5.59.8) seems to ignore all acceleration/speed limits set in configuration (and in reality parts are full of acceleration artefacts, layers slipping due to speed, printer itself is shaking - it's clearly not visual issue), see settings and generated G-code:
[image]
Attaching also 3MF file + printer config. 

Earlier i used "SuperSlicer_2.5.59.6_win64_231231" and had no issues with speed/acceleration.

### Project file & How to reproduce

[cube.zip](https://github.com/supermerill/SuperSlicer/files/14328554/cube.zip)

### Version

2.5.59.8

### Operating system

Win10

ABLomas, 2024-02-19:
I also tested "Also emit limits to G-code" and it seems to fix issue, at least speed is very much reduced and no more uncontrollable shaking.
Two questions remains:

1. why this happens with new version (older did not required this)
2. tooltip says that values are respected in g-code generation, but this is clearly not the case (as required speed values in g-code are far above set in config)

supermerill, 2024-02-21:
acceleration are used as max, but there's nothign for feedrate, true.
I think it's an oversight and should be fixed