supermerill/SuperSlicer #4143: Seem notch doesn't work

Fuechsl, 2024-02-21:
### What happened?

I have applied 0.6mm of seem notch to a huge semi sphere, but I don't see any effect. SS 2.5.59.8 on MacOS.

[image]

[image]
 

### Project file & How to reproduce

[Prj Halbkugel 2024.3mf.zip](https://github.com/supermerill/SuperSlicer/files/14357342/Prj.Halbkugel.2024.3mf.zip)

### Version

2.5.59.8

### Operating system

MacOS 14.3.1

### Printer model

Voron V2.4

supermerill, 2024-02-21:
Also, you reduced the max angle to 90, so it can't do anything.
I'll set its min value to 180, as below 180, it filters out everything.

Fuechsl, 2024-02-21:
Yes, I thought, maybe it's because of the angles close to 180°, that's why I tried to force it by using 90° threshold. I also tried the option "for everything" (because I thought SS may not recognize my perimeter as "round" because it's larger), but this didn't work either. I've tried to print big round things before and had a lot of these ugly zits, so I thought, maybe this new option could be a way to reduce that.
For me it doesn't matter if it's an outer oder inner angle as long as it works as expected. ;)
Thanks for all your work btw.! Is there a way to contact you directly if I have questions or comments that better don't go in here? Discord?

supermerill, 2024-02-21:
with the current, release, if you set "for everything" and put 360 in the angle, you should see it.
Or wait the next nightly.

closed 2024-05-21 by supermerill