supermerill/SuperSlicer #4145: First layer infill differs from next layers along Z axis

inovatorius, 2024-02-22:
### What happened?

First layer infill:
[image]

The next layers:
[image]

### Project file & How to reproduce

[Shape-Cylinder.zip](https://github.com/supermerill/SuperSlicer/files/14366493/Shape-Cylinder.zip)

supermerill, 2024-02-23:
because of extrusion width
It ensure it has the good coverage for the current extrusion width, and that shift everything.

I can add a setting to make it "aligned", (so it will make the infiil% wrong but for one extrusion width)

supermerill, 2024-02-28:
duplicate of #4131

closed 2024-02-28 by supermerill