supermerill/SuperSlicer #4146: Two options to set the fan speed that needed to be set both

inovatorius, 2024-02-22:
### What happened?

So, default values for the fans are 100%
[image]
[image]

If you set "Default fan speed" to 50% nothing happens nor if you set "Max fan speed" to 50%. You have to set them both to 50% in order to fan speed to take the effect event though all the other _speeds_ are set to -1 which should just use the "Default fan speed" (at least that is what description tooltip says)
[image]
[image]

### Project file & How to reproduce

[Shape-Cylinder.zip](https://github.com/supermerill/SuperSlicer/files/14366581/Shape-Cylinder.zip)

jbeima, 2024-02-23:
Unless I am missing something here. I believe this is working correctly. Let me share my mindset on this one.

First the "Max Fan Speed" that you are point to applies to layers that fall under the "Short Layer Time". Since I would believe all of those layers are printing in under 100 second. They would all be classed as short layer times, thus the fan would run at the "short layer time" - "Max Fan Speed".

You can test this by changing that short layer time - time down to say 10 seconds from 100, just to see how it slices, or even 0 to disable it. Just for the sake of seeing how it effects things.

This one can trip people up. Since you also have the "Very Short Layer Time" overrides just below it as well.

:)

supermerill, 2024-02-23:
Current behavior:
 - default speed when possible
 - if below the "short layer time", then it will increase up to the max (here it can't as it's already higher)
 - if below the "very short layer time", it will go to the maximum speed between the current one and the max.
 
 Why?
 because for some feature like bridge, you may want to go higher than any reasonable fan speed, as the layer adhesion isn't important for bridge, as there isn't anything under it.

inovatorius, 2024-02-27:
There is a plenty setting we can adjust and it is a good thing, however looking at the Prusa Slicer UI for the Cooling options it is more obvious what you can change and enable or disable:
[image]

That "Enable auto cooling" checkbox is really doing what I need without me needing to read all that descriptions in the tooltips that after reading adding a bit more confuss.

Also, it is unclear what options can be disabled for example, this one:
[image]

Near the "approximate seconds" line there should be "(zero to disable)" just like in other options:
[image]

And one thing is these options like 0 or -1. This is getting out of hand. Can we just settle with 0 is disable because sometimes -1 is to disable things or sometimes it is 1.
[image]

I cannot recommend anything at this point, because there is just no consistency and I need to take some time to deep dive further into SuperSlicer to actually provide something insightful. I do understand, that all of this was probably discussed already and there is a clear logic behind.

inovatorius, 2024-02-27:
Other thing to mention, M106 and M107 are being spammed all over the place:
[image]
[image]

supermerill, 2024-02-27:
> And one thing is these options like 0 or -1. This is getting out of hand. 

yes, It's not ideal. I may have to redo the gui & settings to add checkboxes for a proper "disable" instead of these values.

> In this example, if we remove M107 we save around 2000 lines of gcode. (31K vs 33K lines)

Yes, known problem. Currently, the fan_mover should post-process the output and remove any unnecessary fan command.
I didn't always activated yet because even if I added many test & many debug I'm not 100% confident it's empty of bugs, but i'll do in the 2.7