supermerill/SuperSlicer #4150: seam :) superslicer vs orcaslicer

godbnc, 2024-02-24:
### What happened?

[image]
[image]

the seam is to big superslicer vs orca but i dont know why 

### Project file & How to reproduce

[cate lateral 55mm-orca.zip](https://github.com/supermerill/SuperSlicer/files/14394395/cate.lateral.55mm-orca.zip)
[cate lateral 55mmsuperslicer.zip](https://github.com/supermerill/SuperSlicer/files/14394396/cate.lateral.55mmsuperslicer.zip)

### Version

2.5.59

### Operating system

windows 10

### Printer model

voron 2.4

supermerill, 2024-02-25:
you mean, the size of the ball to visualize where the seam is is too big?

godbnc, 2024-02-25:
No dimension of x seam

supermerill, 2024-02-25:
I don't understand

closed 2024-02-25 by godbnc

confuzled-dev, 2024-06-16:
@supermerill they are saying that the white section is larger in SS than it is in OS. For instance, if the OS seem looks 1mm wide, it looks 2mm wide in SS. This may only be in slicer and not reflected in prints

godbnc, 2024-06-16:
> @supermerillei spun că secțiunea albă este mai mare în SS decât în ​​OS. De exemplu, dacă sistemul de operare pare să aibă 1 mm lățime, în SS pare 2 mm lățime. Acest lucru poate fi doar în slicer și să nu fie reflectat în printuri

yes is true :)