supermerill/SuperSlicer #4151: Skirt first layer with always 3 loops more than set in Loops

wildwick69, 2024-02-25:
### What happened?

normally I don't use skirts but now I tried and am surprised that skirt has always 3 lopps more than set on 1st layer, the following have the set count. so eg setting loops 3, skirt height 3 results with 1st layer 6 loops, 2nd and 3rd 3 loops what makes removal of skirt from plate sometimes tricky. haven't found a setting to avoid this?
[image]

### Project file & How to reproduce

select Skirt Loops whatever you like (tried with 1,2,3,4)
set skirt height (tried 2,3)

slicing always creates 3 more loops on only 1st layer (1->4, 2->5. 3->6)

### Version

2.5.59.8

### Operating system

W10

### Printer model

Voron Trident

supermerill, 2024-02-25:
[image]
project file?

wildwick69, 2024-02-25:
sorry...
[test.zip](https://github.com/supermerill/SuperSlicer/files/14397091/test.zip)

supermerill, 2024-02-25:
you have `skirt_brim` (extra lines on first layer) set to 3

wildwick69, 2024-02-25:
pebcac, I'm sorry

closed 2024-02-25 by supermerill