supermerill/SuperSlicer #4152: [Feature request] Make the "layer time goal" slow down printing only infill and  internal perimeters.

Murrdo, 2024-02-26:
Almost all printers have ripples when printing at certain speeds. Sometimes there are no ripples only at high speeds (for example 100+ or ​​150+). This makes it difficult to print small parts with short layer times. You have to print several pieces, or print “garbage” next to each other to increase the layer time.
At the moment, "Layer time goal" slows down the printing of all types of lines, it would be very convenient if it were possible to slow down only the filling, or sequentially - filling - internal perimeters, external perimeters.
Those. we print the outer perimeters at 100, and the infill at least at 5-20mm/s

supermerill, 2024-02-27:
Currently:
Priority speed change:
 . internal perimeter
 . infill: sparse, solid, top
 . gapfill, skirt, brim
 . support
 . wipetower
Speed change if there is no other way:
 . external perimeter & thin walls
 . ironing
 . internal bridge
 . support interface
No speed change:
 . bridge
 . overhangs

shinanca, 2024-07-04:
@supermerill  OrcaSlicer do the same, but in different way: https://github.com/SoftFever/OrcaSlicer/pull/5148

supermerill, 2024-07-05:
> @supermerill OrcaSlicer do the same, but in different way: [SoftFever/OrcaSlicer#5148](https://github.com/SoftFever/OrcaSlicer/pull/5148)

the different way is a setting to opt-in/opt-out ?

shinanca, 2024-07-05:
> > @supermerill OrcaSlicer do the same, but in different way: [SoftFever/OrcaSlicer#5148](https://github.com/SoftFever/OrcaSlicer/pull/5148)
> 
> the different way is a setting to opt-in/opt-out ?

They do more simply, just fix speed in filament settings, it`s good, but more useful independent max speed reduction for external perimeters in percent.

supermerill, 2024-07-05:
>  but more useful independent max speed reduction for external perimeters in percent.

I'm sorry, but I don't understand.
I can't see any new fields in the screenshot apart the "don't slow down ext. peri..." / "don't slow down outer wall", there is no "percent" in the thread, and the two percent fields in the screenshot already exists in SuSi.

Murrdo, 2024-07-05:
> Currently:
> Priority speed change:
> . internal perimeter
> . infill: sparse, solid, top
> . gapfill, skirt, brim
> . support
> . wipetower
> Speed change if there is no other way:
> . external perimeter & thin walls
> . ironing
> . internal bridge
> . support interface
> No speed change:
> . bridge
> . overhangs

I still think the best solution is as described above - slow down sequentially, slow down the outer perimeter last. Just blocking their speed is too crude a method IMHO.