supermerill/SuperSlicer #4154: Automatically correct the height of the last layer (or two layers) to obtain the correct print height

Den2016, 2024-02-26:
I want to print a 1.9mm high model. The layer height is 0.2mm (or 0.3mm). As a result, I have to set the height to 0.1 (which slows down printing), or accept that the work piece will not be 1.9mm, but 1.8mm. Is there any way to make the last layer (or several layers) to be of a lower height, so that in the end I get the height of the part as it is set in the model? (without my interference of course ))))

neophyl, 2024-02-26:
What's so difficult about using the VLH tool ? Just use the variable layer height tool on the very last layers.  Takes a couple of seconds to do if you want to make it match.

supermerill, 2024-02-26:
I can see the use. I even tried to implement it when I started to tinker slic3rPE.

Den2016, 2024-02-26:
> What's so difficult about using the VLH tool ? Just use the variable layer height tool on the very last layers. Takes a couple of seconds to do if you want to make it match.

Sometimes you just don't pay attention to that kind of nuance. Until the time comes to assemble a thing from printed parts.....