supermerill/SuperSlicer #4155: Estimated Filament Mass Doesn't Account for Extrusion Multiplier

ndanyluk, 2024-02-26:
### What happened?

After using a full spool of Paramount3D ABS (Density = 1.04 g/cc), the value of the used filament in Spoolman is reported as 1106g on a 1kg spool.  If you were to multiply that 1106g by my extrusion multiplier of 0.92 you would get 1018g.  This implies that when Superslicer is generating the estimates of filament used, it is not accounting for this EM value.

Assuming a properly calibrated flow rate, I'd expect these values to more closely align.  The gcode output and other slicing output appear correct - the estimated filament simply appears wrong.

### Project file & How to reproduce

[Example Project with Incorrect Filament Output](https://github.com/supermerill/SuperSlicer/files/14407427/BZI_V2.4-idler_x4.3mf.zip)

### Version

2.5.59.8

### Operating system

Windows 11 Build 22621.3155

### Printer model

Voron V0.2 with MiniSB

ndanyluk, 2024-02-27:
Hi! Sorry I didn't realize that.

What I'm seeing is that in that example, my calculated "Used Filament (g)" is 59.33.

However, the parts are actually 55-56g. That's about a 6-7% difference and lines up well with the 0.93 EM that I have set in this filament profile. This led me to believe the EM was excluded from the calculation.

[image]

supermerill, 2024-02-29:
I think we agree on that.
I'll try to remove it from the statistics.