supermerill/SuperSlicer #4158: 25598 FAN SPEED

julianjc84, 2024-02-27:
### What happened?

Version 2.5.5.9.8
Default Fan Speed is not allowed bellow 35%

### Project file & How to reproduce

Default Fan Speed less than 35% always lists as 35% minimum.

[image]

Slicer view reflects the 35% fan speed
[image]

### Version

25598

### Operating system

win10

Nbeknel, 2024-02-27:
Duplicate of #4035 
@supermerill , perhaps it might be useful to add a tip/note somewhere in the cooling menu mentioning where one can find the `fan_printer_min_speed` parameter?

julianjc84, 2024-02-27:
solved with searching
fan_printer_min_speed
that's very confusing how fan_printer_min_speed is in a totally different section.

supermerill, 2024-02-27:
what about replacing `The minimum speed is 35%` by `the minimum fan speed is enforced by the extruder to 35%` ?

> that's very confusing how fan_printer_min_speed is in a totally different section.

It's to reflect a hardware limitation of some printer hardware (fan can't spin below a threshold)

julianjc84, 2024-02-28:
something like this would have saved me the confusion
[image]

supermerill, 2024-02-28:
Maybe a description line just below the default fan speed?