supermerill/SuperSlicer #4160: Cooling fan turned on when set to 0, by "normal fan speed at layer: X"

DavidCC00, 2024-02-27:
### What happened?

Filament -> Cooling Settings -> Fan Speed Default
Default Fan Speed:  0
Other options: -1
Disable the Fan for the first layers: 4  
Full Speed of Fan at layer: 2

Cooling Fan turns on (intention was to turn off for entire print).  Note: I never filled out fan off/full speed fan settings.  Not sure why full speed was at 2, seems odd, possibly a bug with importing the voron profile (I experienced another bug that set the voron profile to sprinter and not klipper, too).  

If this is working as intended, please add to the tooltip for Default Fan Speed that even when at 0, it can be turned on by Layer height or similar other language, as the current language makes it sound like it'll stay off when set to 0 (at least at default, since Default speed is green/simple, but layer height fan speed is red/expert.

If not working as intended, I'm wondering if having  "Off till 4 > Full speed at layer 2" caused some unanticipated case in if( x < y) type statement. 

I just switched from years on Cura to SuSr, so I've done almost nothing to tweak SuSr past the default Voron imports.  All I did was change the PLA profile's temp, bed temp, and default cooling fan to be suitable for PETG after a reel of successful prints with PLA.  

### Project file & How to reproduce

[KV1h19muard_b_x2-0g-1h19m.zip](https://github.com/supermerill/SuperSlicer/files/14423363/KV1h19muard_b_x2-0g-1h19m.zip)

### Operating system

Windows 10, Klipper with Mainsail RPi3b

### Printer model

LDO kit Voron 2.4r2revC

supermerill, 2024-02-27:
It seems like it's not on the latest version, can you try again with the 2.5.59.8 ?

zheroz00, 2024-03-10:
I set 'Disable the Fan for the first layers: 1000'. This keeps the parts fan off the entire print. Sort of a workaround I guess.