supermerill/SuperSlicer #4161: First layer infill differs from second layer when using no bottom solid layers

Cyondil, 2024-02-27:
### What happened?

I have created honeycomb patterns in blank spaces before ins various items and have done this successfully in the past.  I am now using version 2.5.59.8 and have noticed there seems to be a problem when using infill with no bottom solid layers.  The first layer seems to be offset/a different scale than the subsequent layers.  This seems to happen no matter the pattern type for the infill. 

First Layer in 2.5.59.8
[image]

Second layer and after (not the same):
[image]

Since I have done this successfully in the past, I decided to download an earlier version, 2.5.59.2 and it worked correctly.  Seems a bug was introduced in a later version of the program.  See below for examples:

First Layer:
[image]

Second layer and after (same):
[image]

### Project file & How to reproduce

[350_Front_A_2.5.59.zip](https://github.com/supermerill/SuperSlicer/files/14424168/350_Front_A_2.5.59.zip)

I take the original file and then make it separate objects to apply individual settings to the "insert" pieces.  To the individual insert pieces, I apply no bottom or top solid layers and then apply the infill settings (20% infill and honeycomb).

I included the project files from both 2.5.59.8 and 2.59.5.2

### Version

2.5.59.8

### Operating system

Windows 11

### Printer model

Voron 2.4 with Stealthburner

devsfan1830, 2024-02-27:
Just want to report that I too have this exact problem as I too do this to add a mesh on my Voron skirts. Had to make a new one to fix some keystone fitment, and suddenly this occurs rather than having the infill neatly stack. This issue did not occur for me either on earlier versions. I am also on 2.5.59.8.

julianjc84, 2024-02-28:
The issue is your Extrusions Widths.
Printer Settings: Width & Flow: Default
Printer Settings: Width & Flow: First Layer

Set the Default and First Layer to be the same % as this impacts how infill is calculated and generated.

also Perimeters & Shell : Quality Only 1 Perimeter: On Top Surface will impact the Top layer infill.

devsfan1830, 2024-02-28:
> The issue is your Extrusions Widths. Printer Settings: Width & Flow: Default **Printer Settings: Width & Flow: First Layer** Printer Settings: Width & Flow: Infill
> 
> Set the **Default** and **First Layer** as this impacts how **infill** is calculated and generated. As well has Infill width plays a big role too, this will likely fix your issue.
> 
> also Perimeters & Shell : Quality Only 1 Perimeter: On Top Surface will impact the Top layer infill. ![image](https://private-user-images.githubusercontent.com/14962510/308390530-a86791f4-423d-4c23-9450-f531af196981.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwODU3MjcsIm5iZiI6MTcwOTA4NTQyNywicGF0aCI6Ii8xNDk2MjUxMC8zMDgzOTA1MzAtYTg2NzkxZjQtNDIzZC00YzIzLTk0NTAtZjUzMWFmMTk2OTgxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI4VDAxNTcwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxNDg1MjQ1MTMzYTQ2MjA4MzM3NjEwZWQ5Y2I5ZjkzZDZlZTMwZWVlN2U0NmMyOGEzMTNmNDc5NmE0ZTgzZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.G2iF4KGPo6oubY9clbMBtLUP2xbvfXVgo_em7SabnVU)

Yep, looks like setting widths identical fixes it. I am able to leave the only 1 perimeter on top surface enabled. Interesting that the older version didn't require this. Was that a bug that was fixed?

supermerill, 2024-02-28:
duplicate of #4131 #4145

Cyondil, 2024-02-28:
Thanks.  Since this is marked as a duplicate, I will close this now.

closed 2024-02-28 by Cyondil