supermerill/SuperSlicer #4163: Aligned rectilinear sparse infill with solid infill alternating

MIRAEngines, 2024-02-28:
Hello, 

Maybe there is a way to do what I need but I cant figure it out.

In prusa-slicer there is an aligned rectilinear sparse infill setting - perfect for my application.
In superslicer there is a rectilinear with 90deg increment. This looks like a same thing but sadly this option affects the solid infill also, meaning that I can not get the cross pattern in those layers - this results in the sparse being visible through the top layer as all the layers are in the same direction. 

Is there a setting to have aligned rectilinear sparse while having solid alternate 90degs?

Thx for the replies and sorry if this is obvious but I cant figure the settings out.

Cheers, Milan

julianjc84, 2024-02-28:
I understand what you mean.

Can get the same result and it is harder. You have to add a modifier box over your shape that is smaller, lower than the top infill count you have and then Disable the Alternate Fill Angle on that modifier box.
[image]
[image]

almost seems like need Alternate Fill Angle for the different infill options sparse/solid/top/bottom/bridge.

supermerill, 2024-02-28:
multiple option to do so:
 . uncheck alternate fill angle
 . or set fill angle increment to 90°
 . or set fill angle template to one entry of 0° (or 45°, or whatever)
 you can set them in modifiers, layer range modifiers.
 
 to assign it to only one kind of surface... not here yet.
 there are many settings that could be set for one kind of surface only, but I need to find a way to create a gui taht just don't multiply everything by 5.

supermerill, 2024-02-28:
for myself:
 - change types of the settings to array
 - first index for default value, the others one for sparse, solid, top, bottom, bridge, internal bridge
 - There is a combo box that let you select the current surface (default, sparse, ...)
 - below, have some sub-groups with settings modifying the index selected in the combobox.

MIRAEngines, 2024-02-28:
Hi,

thank you for the interest in the topic.

I think that simple option of aligned rectilinear sparse would suffice to avoid unnecessary option bloat but that can be just because I was using only this option in prusaslicer as my projects work perfectly with it.

Cheers