supermerill/SuperSlicer #4164: Separate bridge_flow_ratio settings for bridge_infill and internal_bridge_infill

Element-01303, 2024-02-28:
HI! 

    Recently I discovered an interesting problem with surface finish on the top layer of a part (V0 Drive Frames) caused by the bridge flow ratio overextruding over dense infill, causing subsequent solid fill layers to "stack" on top resulting in a very poor top surface finish. I was able to resolve this to my liking by reducing the bridge_flow_ratio, but this introduces another problem with unsupported bridges whereas the flow is not adequate to produce a nice bridge. I am making an assumption here, but my guess is that both the bridge_infill and internal_bridge_infill features both use the same bridge_flow_ratio, and if we were able to set separate flow ratios on each feature, we would be able to really tune this behavior for each circumstance. Please note in the images below for top surface, the pictures do not really portray the differance in feel of the surface finish; The 100% is very rough, where the 40% is smooth to the touch, hopefully this is reflected.

Surface quality with bridge_flow_ratio at 100%:
[image]

Surface quality with bridge_flow_ratio at 40%:
[image]

Unsupported bridge quality with bridge_flow_ratio at 100%:
[image]

Unsupported bridge quality with bridge_flow_ratio at 40%:
[image]

supermerill, 2024-02-28:
and also maybe from overhangs.

Element-01303, 2024-02-28:
> and also maybe from overhangs.

When you say this, is there a detection setting from the overhangs setting stack that I am missing that influences this behavior? I don't want to request a feature that is irrelevant if this behaviour can be influenced by something I missed. That said, I am still working on finding a balance between a single setting for this ratio that both provides a good surface finish, and usable unsupported bridge; I just thought it would be a pretty neat way to tune this :)

Great work by the way, been a long time user, and wanted to say thanks for the work you have put into developing this.

PKav89, 2024-02-28:
Hello!

I might get this issue description wrong, but there is option Bridge Flow Baseline under Width & Flow.
By default it is set to "Nozzle Diameter", which does twice overextrusion after sparce infill (because it somehow counts internal bridge like x2 layer height). But everything works well if it is set to "Keep current flow". 

Does this fix your issue?

Element-01303, 2024-02-29:
> Hello!
>
> I might get this issue description wrong, but there is option Bridge Flow Baseline under Width & Flow. By default it is set to "Nozzle Diameter", which does twice overextrusion after sparce [...]

Hi! I have experimented with this setting, however I have found that the flow rates as reported by the slicer are still very high for this operation leading to the upper layers showing artifacts , and when lowered to produce the desired surface quality, lends to poor unsupported bridge formation. here are some slicer flow reports to show the difference between the two:

100% bridge_flow_ratio w/ baseline set to nozzle diameter:
[image]

100% bridge_flow_ratio w/ baseline set to "keep current flow":
[image]

IMO, the flow rates are still too high for a dense infill to meet my quality expectations (artifact-free top surfaces) for Voron parts. That said, a low ratio will produce functional unsupported bridges over small gaps which are present in the parts I am printing atm, so the lack of independent flow ratios has not impacted me (e.g. the need to span a large unsupported gap).

supermerill, 2024-02-29:
> I didn't know the Nozzle Diameter baseline was set to 2x extrusion flow

It depends on your layer height...

Using the nozzle diameter as the baseline let you enforce the same flow, whatever the condition are. Just change the flow% to reduce or increase the extrusion.
Using the layer height makes the bridge flow dependent on the layer height... not very useful unless fr corner-cases or if your filament can bridge with very little flow.
Using the "keep same flow" is a way to say "don't anything special for bridges".

PKav89, 2024-02-29:
> Using the nozzle diameter as the baseline let you enforce the same flow, whatever the condition are. Just change the flow% to reduce or increase the extrusion.

I don't get it... Same flow regardless layer height?
Well, it, kinda, makes sense for external bridges, but why it is applied to internal bridges (over sparse infill)?

supermerill, 2024-02-29:
when you have 10% infill ratio, there is not many differences between the two.