supermerill/SuperSlicer #4169: Scarf joins instead of seams

perfessor101, 2024-03-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
a new way to minimize seam gaps borrowed from wood working - scarf joints

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
minimize seam gaps through scarf joints

**Describe how it would work**
Be as descriptive as your skill level allows you.
nozzle raises from one nozzle height to the next nozzle height at a 15 degree angle (there may be better angle choices)
on each layer

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
tried adjusting seam gap settings but doesn't work as well 

**Additional context**
Add any other context or paint drawing about the feature request here.
https://forum.prusa3d.com/forum/prusaslicer/better-seams/
https://github.com/SoftFever/OrcaSlicer/actions/runs/8096108803

legend069, 2024-03-05:
this is similar to "Wipe while retracting" ?
but it should be added alongside s_seam_position for a new variable "seam_type"

perfessor101, 2024-03-05:
I wouldn’t put it in with wipe while retracting that isn’t a seam type?
In wood working seams are better at angles and this borrows from that. 
instead of a seam abruptly starting and stopping the seam starts at a tapered angle bridging one layer to the next. 
so instead of stop extrusion move up and over then start extrusion move up and over while extruding a smaller amount. 
Looks like it’s in the next orcaslicer and next PrusaSlicer. 

The links have graphics and better explanations than mine.

halofx, 2024-03-17:
As this is the only thread I see on Scarf for SuperSlicer, and it has been 2 weeks since the last post, I think this conversation is behind the recent beta release of Orca that has implemented scarf seams. I have played with this beta feature, and it is amazing! I really hope all the slicers quickly move to implement it! I used this on a cylinder, with filament that shows everything, and the seams just aren't there!

bubmann, 2024-04-30:
I'd like to add my +1 ... I just printed with OrcaSlicer for the first time to test the feature, for a bottle-cap made in PETg with a 0.6mm nozzle on my MK3s+, and the seam is far less pronounced, even though I forgot to reduce the layer height for the thread. I guess printers crossed the precision threshold needed for the feature. please look into it ...

djvj1, 2024-05-31:
Please add this, looking forward to getting rid of my ugly seems!!

Jaernet, 2024-06-16:
> activate `external perimeter first`
> then `vase mode (no seam)`
> then to choose the start/end height, you have to modify your extruder min layer height.
> 
> 
> I implemented that many years ago but on my prusa mk2 and ender3, the results where meh.
> I tested it for some days, printed many cylinders and the normal.seam yields better results evrytime.
> So I let it rot, and I don't really know if someone used it, as I never had any request/complain on it, just some some old questions.
> I wonder if it's still working as intended.
> 
> If they find a implementation that can improve the seam with current printer hardware, I'll merge/update.

I have tried > activate `external perimeter first`
> then `vase mode (no seam)`
> then to choose the start/end height, you have to modify your extruder min layer height.
Has produced very good results on some models but recently in some tries the prints have gotten some sort of twisted almost thread-like high pitch features inside and out, tried on delta and cartesian printers with same result
Cannot isolate what causes this other than the "seam as vasemode"

Br
/Niklas

DizzieNight, 2024-08-26:
Just found out about scarf joints and would be a great addition to remove seams on my prints

nuxeh, 2025-02-06:
Scarf seams would be wonderful, +1

rdrcrmatt, 2025-06-19:
Any update on scarf seams?

nuxeh, 2025-06-19:
/me just started using Orca slicer and haven't looked back, and incidentally nor have I looked back when it comes to using non-scarf seams

supermerill, 2025-07-18:
As prusa added scarf joint in 2.9, these will be added when the merge will be done.
If someone want to port it before and do a pr, then it will be added before.

cryptopsy0, 2025-09-28:
still no scarf seams

TheAndr0id, 2026-01-13:
> "external perimeters first" is scarf ????? Scarf means ramping, not printing a perimeter first without ramping.

I just tried this with 2.7.62-beta2 and it does do a scarf seam, just not over a set difference.  I've only done a couple of test slices with  cylinders but the outer perimeter is done as three passes of "layer_height/3", then moves in and does the inner perimeters as the correct layer height.

Prusa/Orca is doing the same thing, just limiting the distance and only over a single pass.  Is that better or worse than the implementation in 2.7.62?  I haven't tried printing anything yet, but I'm not sure the thinner layers crossing over the "seam" point over and over is going to help.

Since it's doing 3x the external perimeters, the print time does increase quite a bit.  Doing the scarf seam over a shorter distance has less of an impact.  (for example if you print at 0.24 layer height the external perimeters will be effectively 0.08...)

I think I'll wait for the Prusa 2.9 merge (or just fight with Prusa for the few prints scarf seams might help).