supermerill/SuperSlicer #4170: More power to the cooling fan settings

CuredPrusa, 2024-03-03:
**Is your feature request related to a problem? Please describe.**
Not a problem in nowadays software structure, but a general problem in setting the correct cooling along the printing layers and details in layers, especially with sensitive thermoplastic like Nylon. Casual users using PLA for sculptures and toys would not agree that cooling fan is very important tool and has to be precisely controlled. It's much better than nozzle temperature which is slow, thus inaccurate. Unfortunately, cooling fan control is neglected by modern slicers. 

**Describe the solution you'd like**
A) More independence for controls like
1. short layer time
2. external perimeter
3. top fan speed
4. custom M104 at layer height

B) Separate fan cooling control for infill. 

C) All fan settings available in height range modifier. 

**Describe how it would work**
A) Now all mentioned functions cancel each other by hierarchy coded in the software. Those hierarchy levels should be user adjustable, depending on needs. Maybe if I put some  numbers next to the command function, like "1" for first priority, "2" second, and so on. Example: Now short layer time changes the outer perimeter which I want to remain fixed. Another example... if I force fan by M104 at layer height, it should be THE TOP PRIORITY for other settings and cancel them all, not the lowest priority as it is now. Then there should be a reset available to go back to profile settings at certain height. 

B) At larger or dense infills, fan would cool down perimeters to much and jeopardize the layer bonding

**Describe alternatives you've considered**
Painting tool for fan speed, like seams and support have

**Additional context**
Another problem of the same topic mentioned in #3928

legend069, 2024-03-05:
this is the current fan control settings per SS version 2.5.59.7
are you wanting 'even' more control?

[image]

CuredPrusa, 2024-03-16:
Yes. The quantity of various options doesn't mean much if they cancel each other. I tried to be clear in my post. If I wasn't I can explain more if required.

legend069, 2024-03-24:
> Yes. The quantity of various options doesn't mean much if they cancel each other. I tried to be clear in my post. If I wasn't I can explain more if required.

yes please explain what cooling options you're wanting exactly.
as far as im aware they do not "cancel each other"

>A) More independence for controls like
>    short layer time -already exists
>    external perimeter -already exists
>    top fan speed -already exists
>    custom M104 at layer height - do you mean M106 ? M104 disables the fan, this can be done with ```{if layer_height == xxx}M106 S255{endif}; in the "between extrusion role"/"before layer change" areas```
>B) Separate fan cooling control for infill. -already exists
>C) All fan settings available in height range modifier. -explain more please.

CuredPrusa, 2024-03-29:
Thanks legend069 for your interest. What you've commented as "already exists" does not match the idea. I didn't  say "this doesn't exist". I said "more independence". But never mind.  I have realized I made this issue too general and too wide, therefore it might be hard to set a focus on it. I will actually put one by one idea in a single separate issue for each, as soon as I get to it in my upcoming projects, so I'll keep it fresh with related screenshots.  

So, let's simplify and disregard the rest for a while. 

First of all, I was using version 2.4 which doesn't have infill speed control. I see that 2.5 covered that, it feels much better now. Second; I apologize for M104, I had M106 in mind. You see, every time I enter a fixed M106 at certain layer, it lasts until generator introduces another M106 for some variable speeds settings. And that could be quite short time, for example when "short layer time" is enabled. I'd like to have a manual input as a master override - this is a common logic from automation industry. 
And the height range modifier...
[image]
You see, none of cooling fan options is available at height range modifier. 

So, for now, let's sum it up to 3 problems/proposals in this 4170 issue:

1. Hierarchy of manual M106. Is this what you've tried to explain with 

> {if layer_height == xxx}M106 S255{endif}; in the "between extrusion role"/"before layer change" areas 
?
If so , I didn't quite understand :)

3. Height range modifier options.

4. Issue https://github.com/supermerill/SuperSlicer/issues/3928 where it's clear that there is too much cooling because "short layer time" calculates only time spent on the model, not the support as well. 
[image]
[image]

closed 2024-06-13 by CuredPrusa