supermerill/SuperSlicer #4175: Model does not slice

kozross, 2024-03-06:
### What happened?

When I click 'Slice now', there doesn't seem to be anything happening with this project. I don't get a progress bar, nor do I see any result from slicing no matter how long I wait. If I switch to slice view, I get an empty plate.

### Project file & How to reproduce

[does-not-slice.zip](https://github.com/supermerill/SuperSlicer/files/14505414/does-not-slice.zip)

### Version

2.5.59

### Operating system

Arch Linux

### Printer model

Anycubic Photon Mono 2

foreachthing, 2024-03-06:
You'd need to fix your model first:
[image]

Blender has found many, many more errors:
[image]

If you have a clean model, try slicing again.

kozross, 2024-03-06:
@foreachthing - I asked the model creator to do a repair, and that did indeed fix it. Thanks!

closed 2024-03-06 by kozross