supermerill/SuperSlicer #4176: Arachne with non default settings, internal perimeter errors

TheBoojah, 2024-03-06:
### What happened?

I'm on Version 2.5.59.9 nightly # 692 (and in much earlier versions too), where I am getting errors like this:
[image]
What is curious is that I get more errors on the mirrored part (mirror done in SS). The errors move around when changing "Model rounding precision", but there are always some on this model.

Also got this under extruded internal perimeter all of a sudden, but only on the mirrored part:
[image]

It seems to be an issue with wall transition angle, where I like a value much higher than default (10) of 30-40 to really fill in wedges without gap fill. But perhaps there are other settings I have as nonstandard that affects things:
[image]
With wall transition angle of 10 I can find no errors.

### Project file & How to reproduce

[arachneangleerror1.zip](https://github.com/supermerill/SuperSlicer/files/14512776/arachneangleerror1.zip)

### Version

 2.5.59.9 nightly # 692

### Operating system

Win 11

user32rtu, 2024-03-09:
In your model, lines in the void appear in the Arachne generator. The Classic generator does not have them.
But I have models in which these lines are also present in the Classic generator and disappear when the “external_perimeters_first” parameter is activated.

user32rtu, 2024-03-10:
And further. If there are several copies of the model on the table and there are these empty lines between them, then when the copies are shifted relative to each other, the lines may disappear.