supermerill/SuperSlicer #4177: Accuracy of internal dimensions

Quadra600, 2024-03-07:
### Version
Versione 2.5.59

### Operating system type + version
Linux Mint 21.3 Virginia - Kernel 5.15.0-97-generic

### Behavior
I'm trying to configure the slicer as best I can to get quality prints. After spending a week trying to get rid of Ghosting-like ripples on the Y-face only, which turned out to be Vertical Fine Artifacts(VFA) and/or Motor Resonance Rippling (MRR), which were eliminated by increasing the base print speed to 125 mm/s, I am experiencing some problems with the dimensions. The outside is almost perfect. The inside, measuring 23x23 mm, however, I cannot fix. The 'negative' compensation values it takes. The positive ones seem to ignore them altogether. 
Can anyone verify this?

[image]
[image]

Also, I am having some difficulty at the corners, especially when there are holes. There seems to be some over-extrusion, but I can't figure out how or if it can be compensated for somehow. 

The printer is proprietary, I'm currently printing PLA at 210° on Forex (cold plate) as the print bed needs to be completely redone. 

Thanks

#### Project File (.3MF) where problem occurs
[Test.zip](https://github.com/supermerill/SuperSlicer/files/14528027/Test.zip)

Quadra600, 2024-03-09:
> is this an issue specific to super slicer?
I am trying now with Ideamaker at 60mm/s (with superslicer I print at 120mm/s to avoid VFA problems)

> Have you ever had dimensional accuracy with this printer?
I have made some improvements over the past months and have never had any dimensional problems with Ideamaker. Now I am having some with SS. But Ideamaker I used to use with the old X-axis. Now with the new one I almost always use SS.

> Have you read, reread, and completed Ellis' tuning guide?
Not yet (I think), but I definitely will!

Quadra600, 2024-03-09:
The problem of the internal dimension is also present when printing with Ideamaker.

dougshell, 2024-03-09:
Then this seems to likely be an issue with you printer or klipper
configuration

Quadra600, 2024-03-10:
> Then this seems to likely be an issue with you printer or klipper
configuration

I have attached at the beginning the file used for initial testing. I have made other identical ones...several times...and the result is always the same. 
How can this be a problem with my printer if the outside dimensions are correct while the inside dimensions are not?

dougshell, 2024-03-10:
There are several ways to test/address dimensional accuracy. I would recommend reading, rereading, and completing ellis tuning guide, then checking out cnc kitchens latest video on calibration using the (paid) tool from vector3D.

Quadra600, 2024-03-11:
I'm not a beginner. I designed and built the printer I use personally and it has always respected the dimensions.
But that's not the point... because if the external dimensions of a printed object correspond to those of the CAD object, I don't see why the internal dimensions of that particular shape shouldn't coincide.
I printed the same object but with a 20mm central round hole. And it came out perfectly round and with the perfect measurements.

So there are 2 things... either my digital caliper only measures the external measurements well but not the internal ones if they are square, or there is a conceptual error in the slicers.

I'm not asking who knows what... I attached the file to ask if anyone has ever encountered the same problem. Has anyone here tried to print that file or a similar object? It is a cube with a square section hole inside. Nothing special. Made directly in SS.

Dirkkro, 2024-03-13:
This is very comon on  SS, Prusa, Cura.  Kiss Slicer had an option to select if you like inner or outer dimension to be  precise.
  The inner hole is  always half nozzle diameter to small.