supermerill/SuperSlicer #4185: A proposed improvement in Brim creation functionality.

CaillouxFred, 2024-03-13:
I’ve been utilizing SuperSlicer for a while now, currently on 2.4.58.5 and thought of a useful feature that could be added to the Brim creation functionality. 

I print many things with ABS. ABS has a marked tendency to warp. Hence the substancial usage of the Brim feature. I found that ABS can pull very much on the base, so much that in some instances it become very difficult to keep the print adhered on the build plate. A quite large brim is necessary. A normal brim is one layer thick and I have experiences sometimes a brim to dislodge from the build plate and partialy break. If I could print with a 2 or 3 layers brim it would be usefull. And if there was a capability to break the brim from the print right at the peripheral it would certainly be a useful feature.

I propose the implementation of a “zipper” function to the Brim feature.
It would consist of the capability to add a Brim to the print with some indentations to facilitate the separation from the original print.

Furthermore, I also propose the capability to adjust the Brim layers count.
Introducing some empty indentations in the Brim would introduce fragility into the Brim attachment, hence the usefulness of adding more than one layer to the Brim.
I’ve created an STL files from FreeCAD, to show example of the above proposal. See attached image.
I cannot download the STL file but I took a screen capture to show the idea.

These new Brim functionalities could include parameters like :
* Number of Brim layers
* Distance between each "zip" or "emptyness"
* "zip" width

The distance between each zip is where the material would remain.
The zip width would be the emptyness of material to create a weak area.
The zipper would be distributed all around the print peripheral.
All the zips would be spread all around in an approximate equidistant.
[image]

Any thoughts ?

neophyl, 2024-03-13:
The zippering idea I've not seen before here (but it was made on the Prusa Slicer github https://github.com/prusa3d/PrusaSlicer/issues/10128 )  but the brim height suggestion has been made several times.  

For example
https://github.com/supermerill/SuperSlicer/issues/3279 - existing request for configurable brim/ears height
https://github.com/supermerill/SuperSlicer/issues/1996 - same again

git2212, 2024-03-20:
The "zipper" geometry would be more resilient if it were half circles and not square holes... the sharp corners is where tearing would start especially with materials like ABS and Polycarbonate and potentially make the zipper rim less resilient than the solid one