supermerill/SuperSlicer #4187: Z error at first layer  when heating time is long

Dirkkro, 2024-03-13:
The  strating G-code  is not perfect when  long heat up time is happening. 
The stepper  are powered off during  a long preheat preheat
Therefore a missalignment   may happen  at the Z axes  up to 0.1 mm 

To solve this, a new homeing after preheat  solves it

**Describe how it would work**
The printer  will run homing again and adjust again after preheat

**Describe alternatives you've considered**
This might be a problem in the Marlin Firmware ?
I did not find the timout parameter there for the holding time of the steppers

**Additional context**
This is very   clear at my Ender 3 Printer  and my old MendelmMax 

My Script for Troxy XY2
.......
G28 ; home all axis
;G29 ; auto bed levelling
G1 Z50 F240
G1 X5 Y10 F3000
M104 S[first_layer_temperature]; set final nozzle temp
M190 S[first_layer_bed_temperature] ; wait for bed temp to stabilize
M109 S[first_layer_temperature] ; wait for nozzle temp to stabilize
**_G28 z ; home Z axis_** ;   this is the  extra homing after preheat
G1 z0.28
G1 X2 Y10 F3000
G92 E0
G1 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0

neophyl, 2024-03-14:
Why are your steppers powered off during preheat ?  On all 3 of my printers they don't power off and I have zero trouble with z height.
Any z height differences to the platter are usually caused by thermal expansion, the platter itself moves as it warms up.  This is why things like mesh bed levelling are best performed AFTER you have heat soaked your print surface, get thermal expansion out of the way beforehand and make probing more accurate.

Dirkkro, 2024-03-28:
I verified it, I am not sure why this is happening on my  Troxy XY2  aand on my Ender 3 (first model)  maybe it is  doe to Marlin  Firmware?
If this is not a common problem, then  I  know how to solve it for me. Thank you for answring!