supermerill/SuperSlicer #4192: Feature Request: Recursive Gap Fill

CCS86, 2024-03-14:
I know that many people think that Arachne is "the way forward" but I am not so convinced. I find many examples where it does not work optimally.

- It tends to inhibit arc fitting
- It can increase print time
- It can decrease quality

 I think that refining the classic perimeter generator can provide better overall results, if you don't need super thin features.

The biggest room for improvement in my eyes, would be to add "recursive gap fill" and disallow "perimeter overlap" which is currently allowed (correct me if I am wrong) 80% overlap in external perimeters and 20% in other perimeters. These overlap amounts are not flow corrected. So, in areas where there are long runs of parallel perimeters, can cause massive over-extrusion problems. 

Superslicer  added settings to control maximum allowed perimeter overlap, so that helped avoid over-extrusion. But often this just force a gap fill so wide, it could not be printed with correct fusion. The classic perimeter generator must work in complete loops (or pairs of extrusions). So, if we disallow un-compensated overlap, we inevitably find ourselves with a gap that is just shy of double the defined extrusion width:

[image]

Then, the current gap fill behavior is to fill that with a single extrusion:

[image]

I think it is relatively obvious that this will not be a successful extrusion. Even if capped by maximum volumetric rate, the kinematics of trying to flow plastic rapidly into that form factor just won't work. You will most likely under-extrude that bead, failing to bond on either side, cause excessive nozzle pressure, extruder motor heat, etc.

Instead, if we define a maximum gap fill extrusion width, and ensure that the gapfill is appropriately adjacent to the last perimeter line, we can fill every gap at, or below, this maximum gap fill width:

[image]

Then, if we run the gap filling algorithm one more time, we will get all remaining gaps:

[image]

This seems like the easiest path towards full control of extrusion width, while filling all gaps, and with no defined over-extrusion to "cheat" our way there.

CCS86, 2024-04-26:
Any thoughts on this @supermerill ?

CCS86, 2024-05-04:
I am really struggling with these parts which have converging perimeter lines:

You can see that I have flow tuned so that there is a global slight under-extrusion, but still these converging areas over-extrude badly:

[image]

Using Arachne makes these converging areas worse.

But switching to classic makes other areas worse, as you have to choose between an impossibly wide gap fill line, or uncompensated perimeter wall overlap.

supermerill, 2024-05-15:
currently working on merging the gapfill to adjacent periemter (from & up to a threshold, to have big enough & not too big gapfill).

Your idea can work. It's a bit complicated, but not that much to code. You can even make them loops, like perimeters. I guess it's an idea for big gapfill. If I still have issues with gapfill, I may implement it in the future.

CCS86, 2024-05-21:
> currently working on merging the gapfill to adjacent periemter (from & up to a threshold, to have big enough & not too big gapfill).
>
> Your idea can work. It's a bit complicated, but not that much to [...]

Thanks for the reply @supermerill . I look forward to these gap fill improvements.