supermerill/SuperSlicer #4198: [Feature request]

Nesquirt, 2024-03-20:
It would be nice to add PETG in filament list instead PET, or adding over PET.
It would fix the spoolman plugin advice "The material of the selected filament does not match the selected material in the slicer. Are you sure you want to continue?".
I tried after slicing the stl, to manually change from PET to PETG in gcode and the advice disappear
Thanks in advice mate

sboehle, 2024-03-20:
Hmm, isn't it possible to enter it manually?
[image]

When you don't use the dropdown, but click in the field and enter "PETG" - will this be removed or corrected in any way?

Nesquirt, 2024-03-20:
> Hmm, isn't it possible to enter it manually?
> 
> [image]
> 
> 
> 
> When you don't use the dropdown, but click in the field and enter "PETG" - will this be removed or corrected in any way?
> 
> 

Uhm i dont think it keep the value stored in gcode,ill give a try this evening!

anonymousaga, 2024-04-30:
The value isnt stored. When you type in a custom value, it doesnt let you save and it reverts back to the old choice.

anonymousaga, 2024-04-30:
Something that does seem to work is going into the filament's .ini file and editing it to a custom value. For macOS the ini files are at ~/Library/Application Support/SuperSlicer/filament/

anonymousaga, 2024-05-01:
> I can confirm that changing manually the value to PETG doesn't seem to be correctly stored, as the GCODE generated still shows "PET" instead of "PETG". This becomes a problem when using something like Spoolman to manage spools, which tries to match the `filament_type` with the inventory of filaments in it's database.

Try editing the ini files directly. For me editing the ini files directly worked.

sboehle, 2024-05-01:
Very strange. I tried it again with the latest sc version. I entered "abcdef". It was saved in the filament preset and the generated gcode contained "abcdef" as filament_type.

Nesquirt, 2024-05-03:
> Try editing the ini files directly. For me editing the ini files directly worked.

FIXED WITH THIS COMMENT

closed 2024-05-03 by Nesquirt

AlejandroRivera, 2024-05-03:
> Try editing the ini files directly. For me editing the ini files directly worked.

Yes, i had done this, but every time i saved something in the filament section, the PETG type was replaced with PET :/