supermerill/SuperSlicer #4200: slicer keeps blowing up the scale size 100%

JohnRabbid, 2024-03-23:
### Version
_Version of SuperSlicer used goes here (help->about)_2.4.58.5

### Operating system type + version 
_What OS are you using, and state any version #s_ Windows 10 (build 19045), 64-bit edition

### Behavior
* _Describe the problem_ when slicng it blows up the part to over 100% stock size
* _Steps needed to reproduce the problem_ import part and slice
* _Expected Results_ 1 to 1 scale
* _Actual Results_ 1 to 100 scale
  * _Screenshots if useful_

_Is this a new feature request?_ no
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[Ender_5_X_Stepper_Cover_Chain_Guide.zip](https://github.com/supermerill/SuperSlicer/files/14730055/Ender_5_X_Stepper_Cover_Chain_Guide.zip)

Nbeknel, 2024-03-23:
In the filament preset the shrinkage value is set to 10%, which scales the object by a factor of 10 (= 100% / 10%) in the xy-plane.

JohnRabbid, 2024-03-23:
> In the filament preset the shrinkage value is set to 10%, which scales the object by a factor of 10 (= 100% / 10%) in the xy-plane.

Hey thanks!!!

closed 2024-03-23 by JohnRabbid