supermerill/SuperSlicer #4204: Feature Request: Move "Acceleration control" from "Print settings" to "Printer settings"

PKav89, 2024-03-28:
_Acceleration control_ is now in _Print Settings_. But accelerations are printer-specific. Some printers can do 1500 mm/s2, some can 10000+ mm/s2. So, I guess, it would be better to move _Acceleration control_ to _Printer settings_ and adjust accelerations according to printer profile, not slicing profile.

Nighthawk3D, 2024-03-28:
It is technically in both locations. The printer settings itself has the maximum acceleration values defined for that printer under "Machine Limits". The print settings are defining out the more detailed pieces like maximums during different print operations. Walls, travel, etc.

The printer machine limits are the maximums. So if the print settings say 800mm/s for travel, but the printer machine limits can only move 500mm/s it will only move at that 500mm/s maximum.

You may need to tweak some of the print settings for a particular printer. I usually just copy it to a new preset and append the printer name to the print setting.

I have different print profiles for the same printer depending on different material types as well. Not sure there would be a benefit to organizing it like you are suggesting. It would just cause a large number of printer profiles and still needing matching print profiles on top of that.

This is the same methodology that all Slic3r based slicers (Prusa Slicer, Super Slicer, and Orca Slicer) use.

Arthur-de-Partuur, 2024-03-29:
I use my **print** acceleration (so NOT **printer** limits) also for quality purposes. Speedy prints have also higher accelerations. Quality prints are slowed down. Suggest to leave it "As Is"

PKav89, 2024-03-29:
But different printers have different acelerations to produce "quality print" and "speedy print".

Murrdo, 2024-03-29:
> But accelerations are printer-specific. 

No, different plastics can be printed at different accelerations (for example, very soft TPUs require no more than 750-1000 so that the extruder does not “chew” them, even if the printer can do 15000)
Also, matte plastics can be printed at higher speeds than glossy ones (i.e., the echo is less noticeable on matte ones), which also requires a separate printing profile and not a printer.

CuredPrusa, 2024-04-26:
Since we're at it, I think that retraction should be moved from Extruder to Filament. Yes, the extruder is doing the retraction, but it is generally depending on filament quality and temperature of filament. But I'm used to it already, so it doesn't bother me.  I save all three settings separately for each project. It's rare that I use one setting for more projects. Maybe I would duplicate printer settings if it didn't have retraction settings.

Nighthawk3D, 2024-04-26:
> Since we're at it, I think that retraction should be moved from Extruder to Filament. Yes, the extruder is doing the retraction, but it is generally depending on filament quality and temperature of filament. But I'm used to it already, so it doesn't bother me. I save all three settings separately for each project anyway. It's rare that I use one setting for more projects. Maybe I would duplicate printer settings if it didn't have retraction settings.

You mean like in the filament override section that already exists? You have the baseline retraction on the printer profile, and any overrides are in each filament profile should you chose to use it.

This is the same method used in Prusa Slicer and OrcaSlicer too.

[image]

CuredPrusa, 2024-04-27:
:)
Yes.
 I never use overrides so I didn't investigate. Sorry. I guess this renders my comment moot. 
Thanks for the insight.

closed 2024-06-26 by PKav89