supermerill/SuperSlicer #4205: Arc fitting doesn't work properly.

CuredPrusa, 2024-03-29:
### What happened?

In Archane mode I see G2/G3 only in last perimeter, in Classic moce everything is G1.
[ring_25598.zip](https://github.com/supermerill/SuperSlicer/files/14808049/ring_25598.zip)

### Project file & How to reproduce

Please check did I do something wrong or there a problem within software.
Thanks.

### Version

2.5.59.8

### Operating system

win10

### Printer model

FLsun V400 / Klipper

supermerill, 2024-05-15:
The arc fitting is done after the perimter generation, it's just that the current algorithm (i picked it from bambu) failed to create G2/G3 from the paths from arachne.

Will be better in 2.7 (using the algo from ArcWelder)

supermerill, 2024-05-15:
you can increase the arc fitting tolerance. By putting a higher number, I can have only G2/G3 with your file

CuredPrusa, 2024-05-16:
Yeah, you're right. When I increase tolerance from 5% to 10% I get G2/G3.
Thanks.