supermerill/SuperSlicer #4213: Infill Line Multiplier similar to Cura

phazei, 2024-04-08:
**Is your feature request related to a problem? Please describe.**
In large objects, the infill can take quite a long time to print, and if you lower the infill % the infill can be incredibly weak and flimsy.  By adding a infill multiplier it can double the lines, so similar to how external perimeters adds a lot of strength to the exterior, adding more infill lines makes the infill much stronger without being as dense.

**Describe the solution you'd like**
An option to "multiply" the infill lines.

**Describe how it would work**
In the settings, one would input a number, and the infill pattern will look like the infill was duplicated right next to itself, taking care of overlaps.

**Describe alternatives you've considered**
Using Cura in circumstances where I need better internal support.

**Additional context**
Here's a description in the settings guide:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/infill_multiplier.md

It was also discussed here with lots of pictures:
https://github.com/prusa3d/PrusaSlicer/issues/7044
Where @profOnno did a quick implementation that was only for Gyroid here:
https://github.com/profOnno/PrusaSlicer/commit/90862360de111571272ebbb7b8e0f89f8f841805

The Cura code for generating it is here:
https://github.com/Ultimaker/CuraEngine/blob/main/src/infill.cpp#L358
It seems to have single method that handles almost all the infill patterns.  It needs to take into account if there is an odd or even number for the multiplier, and if the infill lines connect or not. Due to that it isn't supported on "zigzag".  They disable the option for these cases:
`infill_sparse_density > 0 and infill_pattern != 'zigzag' and (gradual_infill_steps == 0 or not zig_zaggify_infill)`

supermerill, 2024-05-15:
is it only to fill the voids in gyroid?

for rectilinear, you can just setup a big width, it will be quicker & even more strong.

phazei, 2024-05-15:
It's so you can set a really low infill density and use the infil as if it has extra perimeters added to the infill.  Yes we can currently adjust the infill width, but that only gets you to 0.6mm.  With this it could be closer to 2mm if the multiplier is like 3.  Many of the infills would be pretty weak and flimsy if used with too low a density since the intersections are too far apart.

RF47, 2025-05-04:
In case anyone is interested, I implemented multiline gyroid fill in Orca Slicer.
https://github.com/SoftFever/OrcaSlicer/pull/9432

[image]