supermerill/SuperSlicer #4214: The height of the corners is 1mm higher on the finished part.

Termin-X, 2024-04-08:
### What happened?

Perhaps this has been discussed more than once. The plastic supply is different in% of the layer height or mm from the nozzle. It is necessary to understand the cutting - how much is the distance between the lines. With a distance of 0.3 mm, plastic is enough, with 1mm between the lines, plastic needs a lot more. A width of 0.3 for the outer perimeter is rather not possible. Printing with uneven feed in different areas(outer(external) perimeter /perimeter/filling)
I print with a 1mm nozzle. The details are very simple. But each slicer has problems with some sections. When passing 2-3 lines, everything is fine. Where there is a perimeter, or filling, the thickness of the finished part increases.
Experts in 3d models will forgive me - I do the support in advance in the model itself. The result is predictable, does not depend on the software, only your own thoughts.
The problem is in the area circled in red. There is a lot of plastic. Very much. The height of the corners is 1mm higher on the finished part. There is no problem when using another slicer. (But in another slicer, the slicing lines are more frequent/thinner) Changes in the settings of this slicer did not give me 100% of the result. I changed the parameter "solid filling layers" + Perimeter overlap + Bridge lines density. What am I doing wrong? Where can I see the solution to the problem?
Version: 2.5.59.8
Build: SuperSlicer_2.5.59.8
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 7 (build 7601, Service Pack 1), 64-bit edition
Total RAM size [MB]: 68,539MB
OpenGL installation
GL version: 4.6.0 NVIDIA 471.68
Vendor: NVIDIA Corporation
Renderer: NVIDIA GeForce RTX 3060/PCIe/SSE2
TronXY X5SA 400*400*400 +Merlin

### Project file & How to reproduce

[image]
[Git_hub.zip](https://github.com/supermerill/SuperSlicer/files/14901214/Git_hub.zip)
[image]

### Version

SuperSlicer_2.5.59.8

### Operating system

Windows 7

### Printer model

TronXY X5SA lovnois board (modifid)

Termin-X, 2024-04-16:
It is possible to formulate this as a sentence. Two options. For each type (Outer perimeter, perimeter, filling .... ) - make an additional option - plastic extrusion coefficient. The second thing is to show the amount of plastic supplied to each element - this can protect against improper consumption. Please poke your nose at the presets of the Trunk settings 1mm nozzle. Or give a link (which printer is similar in mechanics, and has a 1mm nozzle installation)?!

Termin-X, 2024-05-09:
[Uploading Reductor7.zip…]()

supermerill, 2024-05-15:
my guess:
with 50% ext. perimeter width and 25% layer height, 100% bridge flow ratio with bridge width = nozzle diameter, you have a huge overextrusion on the bridge flow.
You shoudl switch to 'layer height' or 'keep current flow' for your bridge computation.