supermerill/SuperSlicer #4217: Version 2.5.59.7 can not generate seams properly

CuredPrusa, 2024-04-17:
**Bug description**
There are many options to adjust the seam but none is delivering a uniform seam as it was before. Maybe it works on simple models, but on a bit more complex, the seam goes everywhere and can not be tamed by available commands. Not even paint seam. 

**Expected behavior**
Seam in the BACK should be seam in the back, or at least aligned in the back. That's how it always was, that's how Prusa has it still. Now I have to use SS 2.4.58, drop additional settings from 2.5, because seam generator just doesn't work properly. I'm using now 2.4.58.4 and it works just fine. 

!!>> Project File <<!!
Here is the project, you can see that same project gives different seam in 2.4.58.4 and 2.5.59.7.
[test.zip](https://github.com/supermerill/SuperSlicer/files/15008788/test.zip)

**To Reproduce**
Steps to reproduce the behavior:
Select seam in the BACK. You can also see that using cost of angle and travel in 2.5.59.7 doesn't make significant changes. 
Changing to archane gives better results, but that is not a workaround. 

**Screenshots**
[image]
[image]
[image]
[image]
[image]

[image]
[image]

**Desktop (please complete the following information):**
 - OS: W10
 - Version [e.g. 2.3.55.5], Use `Help->About`_
 - Printer model FLSUN V400

**Additional context**
Changing randomly the seam point creates unwanted artefacts on thin wall surface.

CuredPrusa, 2024-04-17:
Here's the slicing in 2.4.58.4:
[image]
This is what I call "seam on the back side" !

supermerill, 2024-05-15:
I fixed it, simplifying the overhangs detection.
Not sure on exactly what prusa did here.

CuredPrusa, 2024-05-15:
Thanks heaps!
Now let's see about that nasty M106 :)

CuredPrusa, 2024-05-29:
II have tried the same project on 
Version: 2.5.60.0
Build: SuperSlicer_2.5.60.0_2024-05-21
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19045), 64-bit edition

and unfortunately it isn't any better .

supermerill, 2024-05-29:
on 59.10 release (an my edv branch, so should be on nightly also):
[image]

isn't what's you wanted?

CuredPrusa, 2024-05-29:
The seam point is fine though, but travels are still messed up. 
On such thin wall entering and exiting are visible, and it's not insignificant. In 2.4 (and Prusa any version) this model is sliced with uniform travels: 
[image]
not so in 2.5:
[image]

supermerill, 2024-05-29:
fixed for next version, that was gapfill that discarded (wrongly) their reversibility.

closed 2024-06-05 by supermerill