supermerill/SuperSlicer #4220: Feature proposal - Automatic Temperature Conditioned by Speed

CuredPrusa, 2024-04-18:
Comparable to the same what we already have: Speed conditioned by layer time, I'm proposing a new feature: Temperature conditioned by printing speed (mm/s). 

This could be very useful to prevent overheating on slow speed slow flow layers. And to prevent high nozzle pressure when speed increases.

legend069, 2024-04-22:
this could be useful 'if' the hotend was able to heatup and cooldown instantly.
but due to the real world time it takes to heatup and cooldown adding this feature would cause print issues that can't be fixed with a slicer/firmware

CuredPrusa, 2024-04-23:
Yes, there would be occasions where this automatic temp control could be tricky, but I can see it really helpful when certain printing segments get speed change by manual input or by automatic calculation of layer time. When speed is higher, more temperature is needed, when it goes slower, the temperature should decrease to prevent overheating at lower flow. Since the usefulness would be limited to steady speed of more than several layers, this function should be selectable from the UI. However, I agree, if majority see it not interesting in their projects, who am I to ask for more.

legend069, 2024-04-26:
It would be more beneficial to upgrade the hotend/meltzone rather than increase the temperatures.
There is consequences for printing **TOO** hot
The material begins to break down the hotter it gets. And the parts will be considerably weaker along with other side effects.

The only way I could see this being useful is when super slicer has support for adaptive layer heights for extrusion roles

For the users that can't upgrade their hotend/meltzone I would recommend to use the "inbetween extrusion role gcode" and configure that to suit their requirements.

CuredPrusa, 2024-04-27:
Ok, now that you wrote that, imagine the parameter: "slow_down_below_layer_time" slows down your print for 20 more layers and you're still having the same nozzle temperature. Speed reduction was done automatically but the temperature remains. You get too much melting and stringing.

CuredPrusa, 2024-04-27:
I'm not saying this is an important feature. There are available workarounds. This is just an idea for the time when most of bugs SS suffers from at the moment would be solved.

supermerill, 2024-05-15:
I saw some nozzle with induction heating, if someone has a printer with that, this may maybe be useful.
Imo, it's better to just apply more pressure, and using the hotest temp possible (that won't string).

I guess it's possible to create a range of temperature in the filament setting, and then a speed threshold for the "min speed for using hot temp", and use the right temperature for the highest speed available in the next minute. Quite the hassle to code for such a little effect, imo.

CuredPrusa, 2024-05-15:
I agree, thanks for replying, It's nice to see you on a the run, sweeping the reported issues. 

Keep focus on the big problems and bugs. 
I still find 2.4.58.5 as my favorite, most reliable version, although some of my projects can not be printed without archane in 2.5. 

Thanks for your effort by all means!