supermerill/SuperSlicer #4222: Add Wayland Support

hughesjs, 2024-04-18:
wxWidgets 3.2 [supports Wayland](https://www.phoronix.com/news/wxWidgets-3.2-Released).

Unfortunately, PrusaSlicer (and therefore superslicer) seems to be stuck on 3.1.4.

Could you consider migrating to 3.2?

hughesjs, 2024-05-03:
@supermerill - I'd be happy to do that testing if it builds okay. Not sure what PrusaSlicers plans are

hughesjs, 2024-09-26:
@pacho2 - Have you got a fork/branch I could pull to test?

pacho2, 2024-09-28:
I tried to recompile 2.5.59.10 version with wxGTK 3.2 and it still seems to work (at least with the actions I have tried)

Tapchicoma, 2025-02-02:
I also confirm that `2.5.59.13` release compiles and runs ok with wxGTK 3.2.

There is a relevant PrusaSlicer issue: https://github.com/prusa3d/PrusaSlicer/issues/8284 where I found out that `GDK_BACKEND` is forced to be always `X11` ([here](https://github.com/supermerill/SuperSlicer/blob/e2eef1da2d395d92ad0875c478bca67c0bb8ec9f/src/PrusaSlicer.cpp#L77)). Will try to remove it and check if SuperSlicer works ok in wayland mode + wxGTK 3.2.

eternal-sorrow, 2025-03-20:
Any news on this?