supermerill/SuperSlicer #4223: Bed heating at the end of the start gcode

StrDa121, 2024-04-19:
Hi,
My start gcode is executed before the bed heats up to the target temperature. The nozzle is heated first and then the bed.
The same start gcode don't cause problems in prusaslicer.

Version 2.5.59.8
win10
Creality ender 3 v3 KE

dougshell, 2024-04-19:
Post your start code.

StrDa121, 2024-04-20:
Start code:
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X-2.0 Y20 Z0.28 F5000.0 ;Move to start position
M109 S[first_layer_temperature]
G1 X-2.0 Y145.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X-1.7 Y145.0 Z0.28 F5000.0 ;Move to side a little
G1 X-1.7 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 E-1.0000 F1800 ;Retract a bit
G1 Z2.0 F3000 ;Move Z Axis up
G1 E0.0000 F1800

legend069, 2024-04-22:
can you please upload your slicer project files and a sliced .gcode file
thank you.

StrDa121, 2024-04-22:
Here are the files:
[Cylinder.zip](https://github.com/supermerill/SuperSlicer/files/15065897/Cylinder.zip)

> can you please upload your slicer project files and a sliced .gcode file thank you.

Dirkkro, 2024-04-23:
Somewhere in the start G-Code  you should find:
 
"M140 S[first_layer_bed_temperature] ; set final bed temp"

In my configuration it is placed after homing

Dirkkro, 2024-04-23:
Maybe you did not mark the "Bed"  when you made the  new configuration for the printer

legend069, 2024-04-23:
> Here are the files: [Cylinder.zip](https://github.com/supermerill/SuperSlicer/files/15065897/Cylinder.zip)
> 
> > can you please upload your slicer project files and a sliced .gcode file thank you.

with your current configuration the slicers "start g-code" **does not** contain any commands to heat the bed
the only place it gets inserted is on line ``889``(after your 'start g-code') and ``1543``(other layers target)
you need to add ``M190 S[first_layer_bed_temperature]`` in the relevant location in the "start-gcode"
i would also recommend ticking this box, after you have fixed the slicers "start g-code" so it will use the klipper macros.
[image]

just an FYI, you shouldn't have these sections filled in how they are.
for klipper you should be using the macros, your slicers "start g-code" box should contain something like below and your klipper macros should be configured to accept the slicers parameters 
```
print_start HOTEND={first_layer_temperature[initial_extruder] + extruder_temperature_offset[initial_extruder]} BED=[first_layer_bed_temperature] CHAMBER=[chamber_temperature] MATERIAL=[filament_type]
```

[image]

StrDa121, 2024-04-24:
Thanks. I added M190 S[first_layer_bed_temperature] to my code and it works.

commit 576a86d4 (2018-07-30): Consider filament start gcode when deciding whether or not to prepend heating gcode.

Fixes #4223.
closed 2024-04-24 by StrDa121