supermerill/SuperSlicer #4231: Please clarify duplicated controls (overlap, hole size)

CuredPrusa, 2024-04-28:
In version 2.5 there are (at least) three instances to set the perimeter overlap. Could someone define the hierarchy and differences of those similar commands? Which setting cancels which? I'm aware that some are more defined per perimeter, but what if they're used simultaneously, and should that be the case?

[image]
Let's say this one is a threshold for a thin wall generator. (What does "anti-hysteresis" mean?)
[image]
This one could also be a threshold for a thin wall generator.
[image]
What's the difference between 
perimeter_overlap
and
perimeter_max_overlap

Also, I can see the hole size calibration has (at least) two instances. Moreover, they still work quite erratically. Sometimes it works with minus, sometimes with a plus, sometimes not at all. 

[image]

Should I use just "inner" or "vertical hole shrinking" ? What if I use both? 

Maybe I could test and figure out by myself after some time, but I hope someone familiar with SS code could answer in a heart beat. Thanks!

supermerill, 2024-05-03:
. the first one (overlapping perimeters) allow perimter to be on top of each other when the walls are too thin.
. the second one, allow you to adjust the difference between overlap and spacing. With 100%, it consider that spacing  = square extrusion, and 0% spacing = width  = round edges (see width help/details button).
. the filament max line overlap, it just a threshold for the other ones (just discussed above)

for the compensation:
if (-) then, the compensation will shrink your object, removing matter, growing holes.
if (+) then, the compensation will grow your obejct, adding matter and shrinking holes.
(and if it's not, then I again was fooled by  wrong comprehension)

> Should I use just "inner" or "vertical hole shrinking" ? What if I use both?

'inner' works for all "holes".
'vertical holes' works only for round ones. That reminds me that I should change the label to something like 'vertical round holes'. Note that it doesn't work when there is a "horizontal hole" as this one break the "hole", making it the external wall.

CuredPrusa, 2024-05-11:
> for the compensation: if (-) then, the compensation will shrink your object, removing matter, growing holes. if (+) then, the compensation will grow your obejct, adding matter and shrinking holes. (and if it's not, then I again was fooled by wrong comprehension)

In 2.5 version I have to use minus to enlarge the hole.
In 2.4 version I have to use plus to enlarge the hole. But then the minus doesn't work, doesn't show any movement of the inner perimeter. 

Here is a simple model which I use to test how does the slicer behave. 
[calibration_test.zip](https://github.com/supermerill/SuperSlicer/files/15282849/calibration_test.zip)
And often I have to use either "inner" or "vertical holes" because for strange reason some models don't react predictable to such adjustment. 

It saddens me that such important adjustment is so unreliable.

CuredPrusa, 2024-05-11:
Dear Super
Please open this project in 2.4, zoom into the grid so you can see changes in inner perimeter of the object.
And then change "inner" parameter in following values and observe movement of inner perimeter for each value. 
0
-0,1
-0,2
+0,1
+0,2
+1,0
-1,0

[image]

If that is not a bug, I don't know what is! Please check if 2.5 suffers from the same, because I have noticed some strange behaviour there as well, but I didn't define it yet.

[2-V_FMT2_RING_ES-PLA-ST_SS2.4.zip](https://github.com/supermerill/SuperSlicer/files/15282952/2-V_FMT2_RING_ES-PLA-ST_SS2.4.zip)

CuredPrusa, 2024-05-11:
I have also noticed that common configuration folder for all versions is a mess. When I open 2.4 and 2.5.59 and 2.5.60,  the SS becomes unusable and I have to delete configuration folder. Is there a way to isolate configuration folder for each version?

CuredPrusa, 2024-05-11:
For project [2-V_FMT2_RING_ES-PLA-ST_SS2.4.zip] in 2.5 is the same behaviour as in 2.4. 

Here is how it goes for set point of parameter xy_inner_size_compensation:

 set:     0   +0,1   +0,2   +0,3   +0,4
real:  0.0   -0,2     -0,1     0.0    +0.1

+ = expanding the hole making it wider
- = reducing the hole diameter

CuredPrusa, 2024-05-12:
When I delete config again and run 2.5.59.9, the negative is enlarging the hole. This is getting pretty frustrating. 
I would stick to just one version, to the latest maybe, but there are two reasons (so far) why I return to 2.4:

1.  I have some projects with important custom fan control and until M106 generator is fixed I have no other option but to use 2.4
2.  Gap fill has some real problems with latest versions, it doesn't fill up at some tricky places regardless of changing various setting for gap fill in newer version. It just doesn't fill a small corner while 2.4 is doing it perfectly.

CuredPrusa, 2024-05-12:
PLease make an option for user to chose the option of isolated configuration folder. 
Until then, I'm starting 2.4 from CMD (superslicer-console.exe --datadir settings)
2.4 is my getaway when new version lose control with so many new options buzzing around.