supermerill/SuperSlicer #4235: Force slicer order/direction continuity

DSmakesthings, 2024-04-30:
I print a lot of items that are basically 2 perimeter vases, so both the inside and outside surface are represented as external perimeters.  Most layers print in this pattern:  outside surface, clockwise > inside surface, counterclockwise.

Every once in awhile that changes, and I can't determine a cause.  The order revereses, but direction seems to remain the same.  It goes inside surface, counterclockwise > outside surface, counterclockwise.  This creates very visible ridging in the print surface, and it is especially noticeable using foaming materials like varioshore as the 2nd perimeter printed gets pushed slightly outward.

Proposed solution:  
Force continuity in which surface (inner vs outer) prints first when they are both external perimeters.

How it would work:
I'm not sure how it works, but the new alternate perimeter direction feature makes me think it could be possible and similar to that.

[image]

supermerill, 2024-05-03:
holes perimeters are printed in reverse order.
In a cnc, the cutting bit has to cut in the right side, or it will be pushed in the wrong way (away from the piece instead of chewing into it). when you do your hole, you have to reverse the direction, has you cut in the other "side".
For fff printer, it was kept, even if the extruder doesn't rotate.

There is already some "reverse order" for overhangs. Easy to add... but I may ask you to test it before pushing it.
are you okay about testing on many piece and do an unbiased review of them?

DSmakesthings, 2024-05-03:
Yeah I can definitely test it if needed, thank you for your help and for building/maintaining SS for us all to use.

commit 6dd40091 (2018-07-30): Removed factor 4 from layer height gradation, don't apply min/max layer height limitation after gradation. Fixes #4235 (#4241)
commit 8f26cd76 (2018-07-30): Caught another use of z steps per mm that needs to be clear in its application.

Fixes #4235