supermerill/SuperSlicer #4238: Volumetric Speed in Autospeed not capped

FoldZero, 2024-05-01:
**Bug description**

Volumetric Speed in Autospeed section is not respected and is pushing print volumetric maximum over the threshold set

**Expected behavior**
A clear and concise description of what you expected to happen.
Speed would be capped by either Max print speed OR Volumetric Speed - whihcever is hit first?

[image]

Print on top right shows a max volumetric of 32.510, defeating what has been set in slivcer (30)

supermerill, 2024-05-15:
can you post here the project file (.3mf) (in a zip) for me to test ?

FoldZero, 2024-05-30:
OK, here is an example - seems to be defeatable with the profile i am using CR^_Auto 200MAX - really its a question (i might have misunderstood something) should the lowest speed of Auto speed be respected?
![image](https://github.com/supermerill/SuperSlicer/ass
[max_vol.zip](https://github.com/supermerill/SuperSlicer/files/15504647/max_vol.zip)
ets/12484386/8e699c2c-839b-435c-a8b0-e6708dd2535e)

supermerill, 2024-06-03:
can I have one of your problematic project file to test?

hackthejawn, 2024-09-15:
Hi,

Not OP but having what I believe is the same issue. 3mf file is attached.

From Print Settings:
[image]

From Filament Settings:
[image]

Sliced:
[image]

This is on 2.5.59.13 on Ubuntu.

Thanks.
[GridfinityStorageBox_7x6x6U_Base_test.3mf.zip](https://github.com/user-attachments/files/17007108/GridfinityStorageBox_7x6x6U_Base_test.3mf.zip)