supermerill/SuperSlicer #4239: Mac ARM Build is Broker

aSilentSniper, 2024-05-02:
### What happened?

Sadly the Mac Arm Build is not Working i just get the following error after Moving it to my Applications

`“SuperSlicer.app” is damaged and can’t be opened. You should move it to the Bin.`

### Project file & How to reproduce

Download the ARM build and move it to your Applications folder

### Version

2.5.59.9

### Operating system

macos 14.4.1

JWRu, 2024-05-02:
I have the same issue (Mac mini M2 - Sonoma 14.4.1).
The Intel version is working under Rosetta.

malcinator, 2024-05-02:
I had the same issue yesterday. Found this terminal command in the release notes 

> xattr -cr /Applications/SuperSlicer.app

 and the solved the issue for me on my M1 MacBook Pro.

JWRu, 2024-05-02:
> xattr -cr /Applications/SuperSlicer.app

solved the issue for me, too. (Mac mini M2 - Sonoma 14.4.1)