supermerill/SuperSlicer #4240: fill_aligned_z makes sparsely bridge

shengyan, 2024-05-02:
### What happened?

It seems the new feature ''fill_aligned_z'' in version 2.5.59.9 is making bridge lines sparse in both normal and internal bridges. maybe it should be applied to sparse infill only or turn-off by default.

### Project file & How to reproduce

[image]
[image]

### Version

2.5.59.9

supermerill, 2024-05-03:
can't reproduce, can you send here your project file?

shengyan, 2024-05-04:
It has something to do with "infill extrusion width". Set "infill spacing" to 0.7 for a 0.4mm nozzle, the difference should be visible.
the following project file is based on built-in "0.2mm 0.4nozzle V0" profile, the only change is set "infill spacing" to 0.7mm.
[bug report sample.zip](https://github.com/supermerill/SuperSlicer/files/15208418/bug.report.sample.zip)

supermerill, 2024-05-15:
understood what's going on.
I removed the bridges from the effect of this settings.
Fixed in next nightly & release.

closed 2024-06-05 by supermerill