supermerill/SuperSlicer #4242: SuSi encountered an error - std::bad_alloc

Tau512, 2024-05-02:
### What happened?

App error at the end of slicing, resulting in the following error. Infrequently occurs on custom profiles, but the defaults appear to occur 100% of the time

```
SuperSlicer has encountered an error. It was likely caused by running out of memory. If you are sure you have enough RAM on your system, this may also be a bug and we would be glad if you reported it.
std::bad_alloc
```
plenty of ram available on laptop:

```
$ free -m
               total        used        free      shared  buff/cache   available
Mem:           31800        9701       16313        2309        8554       22098
Swap:           8191           0        8191

```

crashes using any of the default print settings:
[image]

ok with my custom 0.20 profile (note that this was taken AFTER the std::bad_alloc error, with no other changes than print profile selection):
[image]

### Project file & How to reproduce

no project file as it appears to occur on any model.

load STL, change Print profile to a default profile and click slice button.
Error occurs.

### Version

2.5.59.9 Stable

benalleng, 2024-05-02:
occurs 100% of the time on custom profiles for me

darco415, 2024-05-03:
running Linux Mint, same error on both app images

supermerill, 2024-05-03:
can't reproduce with default profile on a basic cube
[image]

Can you upload a buggy project file?
What os do you use?

pfsmorigo, 2024-05-03:
Same here. I get `std::bad_alloc` by using `SuperSlicer-ubuntu_18.04-gtk2-2.5.59.8.AppImage`. Reverted to `SuperSlicer-ubuntu_18.04-gtk2-2.5.59.8.AppImage` and I'm fine now.

Tau512, 2024-05-04:
Fedora20 KDEspin here. using SuperSlicer-ubuntu_18.04-2.5.59.9.AppImage (latest stable), but also occuring on the latest RC ([c0faf97](https://github.com/supermerill/SuperSlicer/commit/c0faf97a5217b01562c46dc8fd326b77d758cf8e)).

I tried with all default presets (print, filament & printer) with a default 'box' shape added. Error still occuring. heres the project file:
[4242project.zip](https://github.com/supermerill/SuperSlicer/files/15208207/4242project.zip)

heres the same model that slices ok and does not exhibit the error with user presets.
[4242project-custom.zip](https://github.com/supermerill/SuperSlicer/files/15208218/4242project-custom.zip)

kipwittchen, 2024-05-22:
This problem is gone for me with the latest version, 2.5.59.10

supermerill, 2024-05-23:
I reproduced it on linux

supermerill, 2024-05-27:
#4258

Routhinator, 2024-06-05:
Happening here on Kubuntu 24.04, AppImage - happens with a custom profile and the nozzle set to 0.2mm

Routhinator, 2024-06-05:
Set the infill from Gyroid > Rectilinear and was able to slice without this error.

Encountered with this STL, which slices on Cura without an issue.

closed 2024-06-12 by supermerill