supermerill/SuperSlicer #4243: Infill not generated when using !0.2mm layer

Tau512, 2024-05-02:
### What happened?

using a layer heigh of something other than 0.2 results in infill not being correctly generated.

0.32:
[image]

0.20:
[image]

0.12:
[image]

### Project file & How to reproduce

Affects any model i've tried. for the example i've just added a 'cube' shape. resize large enough to see infill (i used 175^3 for example)
verify infill generation with 0.2mm layer
change layer height away from 0.2mm.

### Version

2.5.59.9 Stable

supermerill, 2024-05-03:
can't reproduce, can you post here two project file, one "good" and one "bad" ?

Tau512, 2024-05-04:
both requests in the zip. one with 0.2(good), the other with 0.3(broken) with a default box shape. Due to my other #4242 bug, i've had to use my user profiles to generate them:
[4243project.zip](https://github.com/supermerill/SuperSlicer/files/15208240/4243project.zip)

This is on the current latest stable SuperSlicer-ubuntu_18.04-2.5.59.9.AppImage, Fedora40 KDEspin.

Tau512, 2024-05-04:
I'm not sure if i'm jumping on this discovery too quickly but I've doing some more slicing and found that combine infill layers (`infill_every_layers`) appears to be exhibiting the same issue; 1 layer, 31% cubic infill:
[image]

Set it to 2 or 3 and it generates correctly. 
2layers:
[image]
3layers:
[image]

Tau512, 2024-05-04:
I've done more testing with a default box shape primative...
looks like there are 2 issues here.

Theres the issue as originally reported with different layer heights (with example project files).
Theres also the `infill_every_layers` issue where even on 0.2mm layer, it can generate incorrectly.

Tau512, 2024-05-05:
i've tested some more on the RC ubuntu-GTK3 appimages.
issue first occurred on this commit: https://github.com/supermerill/SuperSlicer/commit/3a044453b7deec4c9d19feaab90b96b3df41773d

i understand that new versions may contain new features but when i used my profiles on the https://github.com/supermerill/SuperSlicer/commit/fb26f352922fbdf9a01c8985a1a3f09324b01602 appimage release, then i get the following notification. I assume expected, but just giving you all the info i can incase it helps:
[image]

Nighthawk3D, 2024-05-05:
Also having this issue

supermerill, 2024-05-15:
weird, can't reproduce on windows. I'll try on linux.

justvlade, 2024-05-16:
Oh, I have similar problem under Linux Mint 21 (Ubuntu 22.04)
But I did not think about layer height thingy. I print with 0.32 layer height. And too had trouble, that not always, but on some models it for some reason slice with low % of infill, and ignores percent I choose like 25% 50% or even 100%. It slices like it is 5% or 10% selected. But I had this issue not on every model.
I will try to replicate this too and post here my information later.

justvlade, 2024-05-16:
Ок, here are my two projects of the same model. It uses different profiles. On one profile infill is ok. But on the other infill is strangely very low percent. I could not really understand what difference is. But when I open the 'no' project (in the archive) in 2.4.58.5 version it slices infill correctly. There are two projects the 'ok' and the 'no'. Also screenshots of how they are slice.
Will gladly provide additional information or tests if needed.
[test infill v1.zip](https://github.com/supermerill/SuperSlicer/files/15339954/test.infill.v1.zip)

n3ws, 2024-05-17:
Also having problems with infill.

supermerill, 2024-05-21:
can reproduce with @justvlade project on latest stable.

n3ws, 2024-05-21:
Added my project file if it helps. Stable 2.5.59 on Debian bookworm.

[image]
[infill_very_sparse.3mf.zip](https://github.com/supermerill/SuperSlicer/files/15394443/infill_very_sparse.3mf.zip)

justvlade, 2024-05-22:
I will try to reproduce this on clean VirtualBox Ubuntu then, will try to create some step by step guide or something. Also will check in Windows. May be it is hardware or profile dependent.

Nighthawk3D, 2024-05-22:
I’m on a Mac arm build. Just pulled the latest stable build down from yesterday (2.5.59.10) and it’s still doing it in that build too. If I go back a month or so in nightlies it’s fine.

supermerill, 2024-05-23:
can reproduce on linux

Nighthawk3D, 2024-05-23:
Checked the last 2 nightlies for Mac-arm and I see the same issue, and also not generating top/bottom layers as well if using 3d honeycomb. If I switch to Grid or Cubic it generates top and bottom layers but no infill.

[image]

Here are the two project files.
[PR-4243.zip](https://github.com/supermerill/SuperSlicer/files/15419135/PR-4243.zip)

Let me know if grabbing any kinds of debug logging etc would be helpful in tracking this down.

justvlade, 2024-05-24:
Oh, looks like I am too late = ) But anyway. I installed Ubuntu 24.04 with all defaults, had to install libfuse2 to run 2.5.59.10 AppImage. Chose Creality Ender 3 with 0.6 nozzle. Opened my 'no' project and here we are.

[image]

AlessandroEmm, 2024-05-25:
I have the same issue and same for me,  combining infill set to anything but 1 seems to fix it as well.

supermerill, 2024-05-28:
seems like the fix from @Icenowy #4258 also fix this issue.

justvlade, 2024-06-04:
I checked 2.5.59.11rc by this link
https://github.com/supermerill/SuperSlicer/actions/runs/9370711092
And my "no" project works as intended on both host Linux Mint system and on Virtual Box's Ubuntu 24.04.
Infill generates correctly.

closed 2024-06-12 by supermerill