supermerill/SuperSlicer #4244: Window sizing breaks when opening then closing the Width & Flow help / details dropdown

benalleng, 2024-05-02:
### What happened?

After opening the `Help / details` drdopdown the entire window is forced to a fixed size and cannot be maximized again

[image]

Note that the maximize button in the top left is not present

### Project file & How to reproduce

- Open a project
- go to Print Settings -> Width & Flow
- open the `> Help / Details` dropdown` then close it

### Version

2.5.59

### Operating system

debian 11

### Printer model

creality ender-3 v2

benalleng, 2024-05-02:
this could very well be a problem on prusa slicer also just wanted to point it out

supermerill, 2024-05-03:
yes, it's an issue since the beginning.
After some hours of trying to fix it, I gave up.
If someone want to try to fix it search for "wxEVT_COLLAPSIBLEPANE_CHANGED"