supermerill/SuperSlicer #4246: 2.5.59.9 Linux ubuntu

stsffhome, 2024-05-03:
### What happened?

Application  will not slice ,neither GTK2 or std appimage will complete slicing on any stl's. See attached error message screenshot.
I have tried several files & slicing always fails, all the tested files will slice & print using 2.4.58.5 Gtk2 appimage

### Project file & How to reproduce

[image]
[image]
[baby_yoda_phone_stand_fixed.stl.zip](https://github.com/supermerill/SuperSlicer/files/15199348/baby_yoda_phone_stand_fixed.stl.zip)

### Version

2.5.59.5

### Operating system

ubuntu 22.4

### Printer model

cr10

stsffhome, 2024-05-03:
I must admit I jumped straight from 2.4.58.5 Gtk2, I never saw the 2.5.59.8 release to be truthful

darco415, 2024-05-03:
> ![image](https://private-user-images.githubusercontent.com/11564093/327828149-f7847426-3b17-4c5b-9da4-8c3afab44e93.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ3NjI5MzgsIm5iZiI6MTcxNDc2MjYzOCwicGF0aCI6Ii8xMTU2NDA5My8zMjc4MjgxNDktZjc4NDc0MjYtM2IxNy00YzViLTlkYTQtOGMzYWZhYjQ0ZTkzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTAzVDE4NTcxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIwZjA4ZTJmYzJlNjFlM2I1MzYyMmM3ZWJkZTUzZDYyMTdkMDNjODRhNWNiZGYyNDZhYzljZGZiNTczOTBmMjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jpjJrljFj_uDoKcN_Yx2dgwNp9u9giJvL-oGd8VBjqM)
> 
> I'm getting an error too on ubuntu, this pops up while calculating infill. No problems with 2.5.59.8 or prior.

can confirm also with Linux Mint latest , which I know is Ubuntu fork

Tau512, 2024-05-04:
fyi, the `std::bad_alloc` error is already reported in #4242. we are not alone! :smile:

stsffhome, 2024-05-04:
Not the same error message & outcome as #4242, no memory warning see picture attached to report

Tau512, 2024-05-04:
sorry, i meant to 'reply to' @kipwittchen (and @darco415), not your original reported error!

stsffhome, 2024-05-04:
No worries,I've don't the same thing before today.
It appears there are a few issues with the 2.5.59.9 build.I have installed 2.5.59.8  as Darco415 suggested and it runs ok although compared to 2.4.58.5 Gtk2 the print times after slicing are double ,got a test file printing now that sliced 2hr 36mins with 2.4.58.5 gtk2, 5hrs 32mins sliced with 2.5.59.8, I will see what the quality is when print completed

stsffhome, 2024-05-05:
Further to my last comments re: print time 2.5.59.8 every print fails ,using same printer / filament files as 2.4.58.5
it appears that the z output is incorrect as every print attempt the Z only rises by 0.5 of the layer height+offset.
I have tried 3 different print object STL's and the failure is repeatable.
I will continue with 2.4.58.5 gtk2 until a more stable release is available

justvlade, 2024-05-16:
Phew! So I am not alone! I use Linux Mint 21, which is based on Ubuntu 22.04 I believe. And also had some issues. Well. Not very critical, as I even tried to use latest build (2.5.59.9). But I had no errors, nor crashes.
The issues I had:
- It sometimes does not want to slice infill properly. I set 50%, 100%, 30% and it slices like 5% every time. This is kinda not ok. I did not check how it actually printed though.
- And also I had some trouble with slice preview. Especially when I set inner perimeters fan speed -1 AND outer perimeters fan speed 30. If outer speed is -1 also, it does not lag. But when I set only outer perimeters cooling speed, the preview looks weird, it skips layers on preview and can jump several dosens of em. But the printing is ok. The problem I cannot properly preview slice result. But it can happen in other conditions too. But it has to do something with cooling fan speed settings for some reason.
I can test and provide more info if it is needed.
So I should say I am using 2.4.58.5, too. For now.

darco415, 2024-05-17:
This is weird..so I deleted SS .59..figured I would wait for stable version...in my home directory I see a bin folder? I open it and its a ss gear icon.Clicked on it and 59.9 opens up. I slice a file and no error..prinys fine

supermerill, 2024-05-21:
reproduced, will fix.

supermerill, 2024-05-27:
#4258