supermerill/SuperSlicer #4249: Fan speedup time produces strange results

vovodroid, 2024-05-03:
### What happened?

Fan start is moved too far in some layers:

[image]

Decreasing speedup time from 0.2 to 0.1 makes it even worse:

[image]

### Project file & How to reproduce

[Shape-Box-ss.zip](https://github.com/supermerill/SuperSlicer/files/15202603/Shape-Box-ss.zip)

legend069, 2024-05-04:
this is a tricky one to fix because of the acceleration times are assumed to be instantaneous. so any time based features will always be a bit off till that main issue for calculating time from movement speed is fixed.(ignoring any time it takes for the physical fan to speed up)

vovodroid, 2024-05-06:
Latest code (not released yet) produces better results, so still quite strange sometimes.

supermerill, 2024-05-15:
In this case was putting the fan at the start instead of the end.
Also increased precision for long lines.

closed 2024-06-05 by supermerill