supermerill/SuperSlicer #4250: Feature request: ~~Spiralize~~ Print Outer Contour

DavidCC00, 2024-05-03:
**Is your feature request related to a problem? Please describe.**
There are a number of times I want to print the shell of a model.  At times I've been able to do this with spiralize outer contour/vase mode.  However, it doesn't work on everything, especially bridge areas, or things with multiple top sections like the top of a lego block. 

**Describe the solution you'd like**
I imagine that spiralize outer contour has some way of "skinning" a 3d item.  The principle is to use that "skinning" function to create a shell or surface of the outside of an object.  Then, assuming it works this way, let people print that shell at 1 perimeter width, but unlike spiralize outer contour, it allows for travel moves. 

**Describe how it would work**
(as above) 

**Describe alternatives you've considered**
I've tried to do this kind of thing on my own using Blender, Fusion 360, Tinkercad.  My results are always time consuming and don't always work out.  One of the main challenges is making sure the outside doesn't get too thin or too thick.  For example, an RC car body printed in something like Petg or Nylon, you don't want too thick, because you want it to bend, not break, on impact. 

**Additional context**
A few examples of items this would work for.  Additional modifications might be necessary for models but trivial even using something like tinkercad with little experience (adding/subtracting blocks to remove parts or fill parts).

1. Change solid car models into RC car or toy car shells.  https://www.thingiverse.com/thing:2775107 into (using this since it was actually remixed this way, so you can see kind of the goal).  https://www.thingiverse.com/thing:4182930

In this example, you'd use tinkercad to first remove the tires with some cylinders, then put blocks to make the wheel wells flush with the fenders (which is how RC car bodies come so they're easier to paint).  After painting, then you cut out the wheel wells. 

2. Turn head sculpts into halloween/cosplay masks.  https://www.thingiverse.com/thing:389442 

3. Make Gridfinity models with multiple bases without making the bottom 1cm solid layers.  

A lot of my assumption here is that the legwork of "skinning" a model is already in the spiralize outer contour feature.  If this is tremendously more difficult because that assumption is wrong, I withdraw my suggestion. :)

supermerill, 2024-05-15:
> but unlike spiralize outer contour, it allows for travel moves.

how does it works?
what I can think about:
- When there is some infill to print
- When the perimeter for this layer is printed (so it gets to the layer height)
- then goes over it a second time to print the perimeter to the current layer height (like for vase mode for external perimeter first)
- now that we have the outside wall at our layer height, print the infill.
- for the next layer, start the perimeter, but start from 0 height instead of the full height.

?

trysten, 2024-09-10:
one of the attractive features of "vase mode" is the lack of seam. allowing for travel moves may cause a visible seam. that may be why this kind of spiral-hybrid hasn't been developed yet.
My thinking is that vase mode is only appropriate in specific circumstances, but any non-planar experiments are something we should push forward