supermerill/SuperSlicer #4253: Mac 2.5.59 stable edition arm dmg file is corrupted

Fuechsl, 2024-05-04:
### What happened?

When I start SuperSlicer I get a notification of MacOS that the file is corrupted and can't be started.

### Project file & How to reproduce

No file, startup doesn't work.

### Version

2.5.59 stable

### Operating system

MacOS 14.4.1

anonymousaga, 2024-05-05:
Run this command in terminal:
`xattr -rd com.apple.quarantine /Applications/SuperSlicer.app`

This is a common macos gatekeeper dialog. It occurs because the app is not signed with a proper certificate (which costs money and isnt on Supermill's priority rn).

Please close the issue if this solution works.

anonymousaga, 2024-05-05:
Duplicate of https://github.com/supermerill/SuperSlicer/issues/4239

Please close :)

mszulen, 2024-05-11:
It seem to be, indeed, corrupted. The normal version works and shows 152MB after unpacking. The ARM version shows only 5MB

Fuechsl, 2024-05-11:
Told you so… ;) Rosetta version works though.