supermerill/SuperSlicer #4255: SS dosen't work on my Notebook Win11

ErrorindriveC, 2024-05-05:
After the Super Slicer window opens up, I receive two error messages (See attached images). If I press 'OK', Super Slicer closes.

I'm using the latest version, SuperSlicer_2.5.59.9_win64_240430.

I don't encounter any issues when using my PC (Windows 10).

**Desktop:** 
 - HP Spectre Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz   1.99 GHz
 - WIN 11 Home Version 23H2
 - Craftbot3
[image]

stxve4203, 2024-05-05:
Try downloading NVIDIA graphics driver.

ErrorindriveC, 2024-05-06:
The newest driver is installed. Still doesn't work.

konasimakopoulos, 2024-05-12:
Try moving the superslicer folder to a path without special characters (use C:\Superslicer for example).

supermerill, 2024-05-15:
Seems like win11 put characters in paths that are taggued as "invalid". Maybe because I compiled with win10.
If the os give the slicer the userpath with characters that are recognized invalid by the software, it will quit whatever.
I'll have to install a win11 with a not-english locale to test that.
In the meantime, I fear that the only way to avoid it is to run it from a user that have only ascii character.
You can also try to activate the beta utf-8 support, it helps sometimes.

ErrorindriveC, 2024-05-15:
Thank you very much!!! 
I chanched the Account-Name because ther was a (ü) in it. 
And now it works.
Nice.
Thanks again.

closed 2024-05-15 by ErrorindriveC