supermerill/SuperSlicer #4256: GUI problems (SuperSlicer_2.5.59.9_2024-04-30 arm)

zuhra031, 2024-05-07:
### What happened?

GUI problems

### Project file & How to reproduce

When settings page can be scrolled, it scrolls under the fixed GUI element (it happens on all tabs when settings can be scrolled) - see screenshot

[image]

+ there is a some small artifact in top left corner - artifact is there no matter what tab/screen is open
[image]

### Version

SuperSlicer_2.5.59.9_2024-04-30 - arm version

### Operating system

macOS Version 14.4.1 (Build 23E224) - Mac Mini M1

zuhra031, 2024-05-08:
Is not a big problem - for now.

Ive tried every option - problem is still here
[image]

JWRu, 2024-06-07:
This cosmetic problem is still there.
SuperSlicer_2.5.59.11_macos_arm_240605
Mac mini (M2) - Sonoma 14.5

ipa64, 2025-05-07:
Samething on Mac intel
There is another open issue for the same bug.
[https://github.com/supermerill/SuperSlicer/issues/4581](url)