supermerill/SuperSlicer #4257: Errors when slicing this high quality mesh

thetouchablegod, 2024-05-09:
### Version
2.5.59.5

### Operating system type + version
Linux Endeavour 6.6.30-2-lts
Hardware:
Ryzen R5, 16gb RAM, 3060 6GB

### Behavior
* _Describe the problem_
Improper slicing. Program gives error "SuperSlicer has encountered an error. It was likely caused by running out of memory. If you are sure you have enough RAM on your system, this may also be a bug and we would be glad if you reported it.
std::bad_alloc"

* _Steps needed to reproduce the problem_
-Load this relatively large stl (around 2.5MB) 
-Slice

* _Expected Results_
 - proper slicing
* _Actual Results_
- Improperly sliced with walls missing and errors show.
  * _Screenshots if useful_
[image]
[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[new-hydrant.zip](https://github.com/supermerill/SuperSlicer/files/15262440/new-hydrant.zip)
[image]
[image]

konasimakopoulos, 2024-05-12:
Sliced provided model correctly with latest release(2.5.59.9) on windows 11. 
Could you try updating and see if it is solved?

thetouchablegod, 2024-05-12:
> Sliced provided model correctly with latest release(2.5.59.9) on windows 11. Could you try updating and see if it is solved?

I apologize, this was with the latest release 2.5.59.9. I will check to see if the problem occurs under Windows for me.

thetouchablegod, 2024-05-12:
I can confirm it slices in Windows 11 just fine.

th0mpy, 2024-05-13:
I am seeing the same error on the MacOS version. Also running 2.5.59.9 on MacOS 14.4.1 (Sonoma). 64gb memory, M3 MacBook Pro. 

Any ideas?

th0mpy, 2024-05-13:
Switching to Gyroid to Rectilinear for infill seems to clear the error. 

Another interesting error, if I switch to 3d Honeycomb I get a new error "Coordinate outside allowed range"

thomispt, 2024-05-15:
I've got the same issue slicing anything on macOS ARM, dropping to 2.5.59.8 fixes it

MQ1995, 2024-05-28:
I've got the same issue, but just when slicing some models, not all.
I'm on Arch Linux (EndeavourOS) running the latest appimage. I tried older versions as well.
The particular model i cannot slice is this benchmark model: https://www.printables.com/model/209121-cali-dragon

supermerill, 2024-05-28:
you can try the last 'rc' build (in the [actions menu](https://github.com/supermerill/SuperSlicer/actions), commit 1d7a679, label `fill_aligned_z: fix uninitialized atomic_int64_t (#4279)`)
it's the 59.10, but with a fix that should these weird bugs.

thetouchablegod, 2024-05-28:
Can confirm the latest [rc](https://github.com/supermerill/SuperSlicer/actions/runs/9257368504) and [nightly](https://github.com/supermerill/SuperSlicer/actions/runs/9265610076) fix the issue. At least with my hydrant model. I haven't tried any others reported as giving problems.

closed 2024-06-12 by supermerill