supermerill/SuperSlicer #4260: 2.5.59.8 gap fill doesn't do the job right

CuredPrusa, 2024-05-15:
In 2.5.59.8 I can not find a proper setting to fill the corners with gap fill, marked with yellow arrow

[image]

Version 2.4 fills that corner with gap fill instantly without any special settings. Why?

Here's the project:

[test.zip](https://github.com/supermerill/SuperSlicer/files/15325228/test.zip)

supermerill, 2024-05-16:
put something on min width & min length, or they will use default (which ecude too short gapfill)

CuredPrusa, 2024-05-16:
Thanks. Yes, it works now when I have set min length to 0.1.
But I think the value zero should be related to AUTO or ALL, and not restricting functionality if I didn't ask for restriction.

supermerill, 2024-05-16:
yes, it's auto.
and the auto is too high for your need.
I'm going to change how disabled/enable are displayed in next major version.

closed 2024-05-16 by CuredPrusa