supermerill/SuperSlicer #4264: No retraction / deretraction on 2.5.60 nightly

kymlalu, 2024-05-16:
### What happened?

No retraction / deretraction after upgrading to 2.5.60.0 from last stable relase ( 2.5.59.9 ). 
It seems that it is 3d model indenpendent .
Sliced in 2.5.59.9
[image]

Sliced in 2.5.60.0
[image]

### Project file & How to reproduce

Do it first with older version ( 2.5.59.9 ) and as a second use newer one ( 2.5.60.0 ) as it seems that there is something different in config which stays here even after downgrade from newer version.

To  remove any variable i deleted my SS settings from c:\Users\user\AppData\Roaming\ ( whole Superslicer dir ) and made myself sure that there are no leftovers in other folders. Then after launch i selected one printer ( in my case V0 0.4mm ), opened STL and sliced it. 
So you will have pure default settings with one printer, few basic material choices and print profiles. Don't touch any setting.

Then in sliced view, choose retraction and deretraction visualisation and observe result. Then repeat whole process ( deleting SS folder, adding printer, slicing.. ) and look into gcode.

[generated gcodes.zip](https://github.com/supermerill/SuperSlicer/files/15336909/generated.gcodes.zip)
[project files.zip](https://github.com/supermerill/SuperSlicer/files/15336914/project.files.zip)

### Version

2.5.60.0

### Operating system

Win 11

### Printer model

Voron 0

Hun73rdk, 2024-05-17:
I have tons of stringing in the latest build also i changed to orca slicer to see if it was my hotend that was broken and there i have no stringing at all

Pavulon87, 2024-05-18:
Yes, I noticed this bug too
Temporarily uncheck **Only retract when crossing perimeters** @ Print settings -> Infill

supermerill, 2024-05-21:
I think it's a gui issue, maybe. I can't see any missing retraction/unretraction in the first layer in your gcode.
For other layers, well it's 0.05 layer height vs 0.2 ...

supermerill, 2024-05-21:
checked on another project, and there is a problem after the first layer.

supermerill, 2024-05-21:
nevermind, it's because i had 'only_retract_when_crossing_perimeters' set to true in the beta.
I'll default it to false.

supermerill, 2024-05-21:
found the bug, will be fixed in next nightly

kymlalu, 2024-05-21:
I was writing reply to other message from you here and meanwhile you find it.
I'm glad that i'm not blind or crazy ( and didn't waste your time with my stupidity 😅).

supermerill, 2024-05-21:
I optimized it a lot to speed it up. but when it's on a top layer, as the top surface may be excluded, that breaks some assumptions and so it's too optimized and doesn't check the right stuff.

kymlalu, 2024-05-21:
I just downloaded newest nightly build and according to gcode preview in slicer everything seems to be OK. I'll print something tomorow but i expect it to be good now.

closed 2024-06-05 by supermerill