supermerill/SuperSlicer #4265: Gcode view that shows what the limiting setting was for speed

Tirpitz93, 2024-05-16:
**Is your feature request related to a problem? Please describe.**
I would like to be able to analyse which setting is limiting speed.

**Describe the solution you'd like**
I would like a view option in the GCODE preview tab that colour codes the model/preview with the setting that limited each instruction

**Describe how it would work**
_I am not sure how the slicer works under the hood but would it be possible to tag each move with the setting that the max speed was limited by such as machine limit/filament max volumetric flow/percentage max flow from print settings?

**Describe alternatives you've considered**
naively trying to find the setting that limits speed and changing it to see effect on print time estimate

**Additional context**
Add any other context or paint drawing about the feature request here.

supermerill, 2024-05-21:
It's bothersome. Maybe with verbose...

FoldZero, 2024-05-25:
I want this so bad.  speed is kinda fucky in supersclicer, I would love to know what every preimeter is suddenyl blue adn limited at 30mm/s, when the slicer shows a 250 top speed.

closed 2024-06-12 by supermerill